Atmosphere-NX/Atmosphere #1339: Templatize reboot2payload

noirscape, 2021-01-16:
## Feature Request

### What feature are you suggesting?
#### Overview:
- Load reboot_payload.bin from 2 different paths, with one acting as a fallback.

#### Smaller Details: 
- At the moment, reboot_payload.bin is loaded from `sdmc:/atmosphere/reboot_payload.bin`. 
- This is a file that is by default included in a release of Atmosphere. 
- This causes a sideeffect where, upon updating Atmosphere (by replacing the contents of the SD card with the contents of the zipfile, which seems to be intended behavior when updating, as indicated by developers and support staff on ReSwitched), the reboot_payload.bin file becomes overwritten with the latest version of fusee-primary. 
- This causes a problem when a user prefers to use a different payload as their reboot_payload.bin file (such as Hekate), because this adds a separate step to replace the reboot payload file with Hekate.
- The solution to this would be to load the reboot_payload path from two places, with `sdmc:/atmosphere/reboot_payload.bin` as the default and a different path (for example `sdmc:/atmosphere/default_reboot_payload.bin`) as a fallback if the first file does not exist at the time atmosphere sets the reboot payload path. This would allow users to use a release version of Atmosphere without needing to modify the included parts, even when setting a custom reboot payload.

#### Nature of Request:
- Addition
  - Similar actions were done for setting up and loading the configuration files to launch atmosphere.
  - Hekate currently includes a dedicated configuration option to overwrite the reboot payload, specifically to adress this specific issue.

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
- Usefulness described in smaller details: It would reduce divergence of Atmosphere zipfile releases and reduce friction when updating Atmosphere.

SciresM, 2021-01-17:
`Load reboot_payload.bin from 2 different paths, with one acting as a fallback`

no.

We can just stop distributing a default, though, since fusee is already default-in-code?

noirscape, 2021-01-17:
That would also be a solution I think?

SciresM, 2021-02-16:
I've thought better of this -- if we stop including it in the zip, reboot to payload nro breaks and also people who have the file already have their shit break when we update fusee-primary.

Not really sure how to deal with this, I think the status quo is the best option.

closed 2021-02-16 by SciresM