Atmosphere-NX/Atmosphere #1340: A fatal error ocurred when running Atmosphere

Ruisusenpaii, 2021-01-17:
i need some help,i tried to update the atmosphere version to 0.17.1 and it crashes when i try to start it.

Program ID:0100000000000004
Error desc: std: :abort () called (0xxfe)
Report saved to /atmosphere/fatal_errors_report_0000000000000bc64197.bin

SciresM, 2021-01-17:
Please upload the report binary.

Ruisusenpaii, 2021-01-17:
> Please upload the report binary.

https://mega.nz/file/YhUyzBLb#j6HVpGsDUh1YgtGEkNLb--B7JkaWOly1szfQoO1cA8M

SciresM, 2021-01-17:
Update mission control.

closed 2021-01-17 by SciresM

Ruisusenpaii, 2021-01-17:
> Update mission control.

how?

SciresM, 2021-01-17:
mission control is a homebrew sysmodule you have installed. follow the instructions in its github repo, presumably.

Xronium, 2021-01-20:
dude thank you so much, i've been looking for what the problem was, and mission control has a title id of 010...0db i believe? or something different tho the error i was getting so thank you

a lot

chester4ka, 2021-02-06:
Hi, same error, tell me what to do?
Report https://mega.nz/file/f98EATxa#jOBNB6zuIGPfpCKzDNv7ChWOocthTE58tqeHu5Dl6Z4

SciresM, 2021-02-06:
@chester4ka your error is not the same. You're using a custom sysmodule that needs to be updated to be compatible with AMS 0.18.0.

chester4ka, 2021-02-06:
The error occurs when starting tinfoil.

SciresM, 2021-02-06:
Piracy is not supported.