Atmosphere-NX/Atmosphere #1343: Bypass region lock of Tencent-Nintendo game cards.

Randname666, 2021-01-19:
## Feature Request

### What feature are you suggesting?
#### Overview:
- Bypass region lock of Tencent-Nintendo game cards.
#### Smaller Details: 
- Though literally all Switch game cards are region free, Tencent-Nintendo ones, which are released in mainland China, are region locked to only run on Tencent-Nintendo Switch units which are also released in China.
Inserting a Tencent-Nintendo game card into a regular Switch will cause error 2016-2101 and refuse to run.
SX OS has already done that but not sure how it's done due to its closed-source nature. However, by patching `nn::fs::GameCardAttribute` and / or gamecard header offset 0x190, bit 0x24, it might work (untested).
(source: https://switchbrew.org/wiki/Filesystem_services#GameCardAttribute and https://switchbrew.org/wiki/Gamecard_Format#Gamecard_Info)
#### Nature of Request:  
- Addition
- Bypass region lock of Tencent-Nintendo game cards.

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

### Why would this feature be useful?
- Able to run those game cards without a Switch unit from certain region.
- Able to extract and load back contents from such game cards for archival purposes without the need for Switch units from certain region.

SciresM, 2021-01-19:
`However, by patching nn::fs::GameCardAttribute and / or gamecard header offset 0x190, bit 0x24` no, because this would invalidate the HFS0 tree used to store validated gamecard data, since that derives hmac key from this value. You'd get an abort in FS.

Is there some reason this should be in atmosphere and not a set of community maintained ns-patches?

I'm mostly just not interested in creating and maintaining ns patches for this.

Closing the issue since this seems like the kind of thing where the answer is "community makes region free patches", same with e.g. browser ssl verification disable.

If there's some cogent argument otherwise, continue commenting on the issue. If the argument is "but I want you to be the one to make and maintain the patches", that argument is not cogent.

closed 2021-01-19 by SciresM

Randname666, 2021-01-19:
No more arguments.
But I would like to know if there is a guideline for what Atmosphere project cares and what not in order not to post "definitely wontfix" issues again.
Since "Bypass internet access check" showing in the issue area also seems to fall in the scope of "community makes it rather than Atmosphere".

SciresM, 2021-01-19:
That issue's open because I actually have no idea what one would need to do to solve it, and it may end up being best served as code-not-patches (a mitm?)

It's also low priority, in case my not commenting on that issue in almost a year is indicative to you.

carcaschoi, 2021-01-24:
use Tecent switcher ui