Atmosphere-NX/Atmosphere #1344: Web Applet frozen

thaikhoa, 2021-01-19:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Hi,
I'm using both Hekate and fusee primary, Atmosphere 0.17.1, Fw 11.0.1.

Games with connecting to web applet such as in-app-purchase or terms and conditions will freeze the whole system, force to powering off.

I don't know whether the issue is just me.

Masamune3210, 2021-01-19:
please go back and actually fill out the template

thaikhoa, 2021-01-19:
Thanks, just done editing

SciresM, 2021-01-19:
Does the issue occur using stock atmosphere release zip with no custom sysmodules?

thaikhoa, 2021-01-20:
Without any sysmodules, everything works just fine. any other sysmodules that I've added could lead to the issue.

Masamune3210, 2021-01-20:
If it works fine without sysmodules then it's very likely an issue with those sysmodules or from having too many installed at once, not with atmosphere

AuroraWright, 2021-01-27:
It was explained here i believe: https://github.com/Atmosphere-NX/Atmosphere/issues/808#issuecomment-587184145 . So this is just a duplicate

closed 2021-01-27 by SciresM