Atmosphere-NX/Atmosphere #1346: Switch crashes after being left in sleep mode overnight

Grass0001, 2021-01-23:
## Bug Report

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]
left the switch asleep charging in the dock overnight, upon checking it the next morning it has crashed
[ Did you make any changes related to Atmosphère itself? ]
no

### How can the issue be reproduced?

leave the switch in the dock overnight, it will crash at some point

### Crash Report

https://gist.github.com/Grass0001/ff7b49b38f3503a064124c652339eda9

### System Firmware Version

11.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: official hekate 5.5.3
- Do you have additional kips or sysmodules you're loading: sys-ftp-light, SwitchPresence-Rewritten, custom themes
- Homebrew software installed: [ daybreak, reboot to payload, nxdumptool, simplemodmanager, switch presence, nxthemesinstaller ]

### Additional context?

masagrator, 2021-01-23:
You didn't provide Atmosphere version and what you have used to launch it - fusee-primary or Hekate.

Have you tried to uninstall all custom sysmodules and check what will happen?

Grass0001, 2021-01-23:
> You didn't provide Atmosphere version and what you have used to launch it - fusee-primary or Hekate.
launched by hekate 5.5.3 into atmosphere ver 0.17.1
> Have you tried to uninstall all custom sysmodules and check what will happen?
nope, ill try that later though

fincs, 2021-01-23:
I had this same exact crash in the wlan sysmodule happen to me a month ago or so. At the time I was running HOS 10.2.0 with Atmosphère v0.15.0. I don't use nor have any "custom sysmodules" (or any other similar 3rd party modifications) installed on my system, and I exclusively use fusée to boot Atmosphère. This didn't happen overnight, I merely left my console in sleep mode (unplugged) and came back a bit later; it was sitting there displaying a fatal.

[01606325440_0100000000000016.log](https://github.com/Atmosphere-NX/Atmosphere/files/5860134/01606325440_0100000000000016.log)

Grass0001, 2021-01-26:
after a few days of the sys modules being disabled it has not crashed again, thanks for the help

closed 2021-01-26 by Grass0001