Atmosphere-NX/Atmosphere #1348: Atmosphere Fatal Error when launching game

Wandereule, 2021-01-26:
## Bug Report

### What's the issue you encountered?

Set up New Atmo withh Meso (since my SD Card Wiped. Setup as Emummc) Installed some games some working and others e.g. Spyro crashes aftwer the "toys for bob"-logo and gives me a fatal error (also happen on other games like PixARK. Games like Zelda or Mario kart working like a charm) (also checked that the NSP Files aren't corrupt)

### How can the issue be reproduced?

just booting into atmosphere and starting a game, game starts normal for the first seconds and crashes with atmo fatal error

### Crash Report

[01611173900_0100000000000014.log](https://github.com/Atmosphere-NX/Atmosphere/files/5873821/01611173900_0100000000000014.log)

[01612170250_0100000000000014.log](https://github.com/Atmosphere-NX/Atmosphere/files/5873778/01612170250_0100000000000014.log)

[01611442625_0100000000000014.log](https://github.com/Atmosphere-NX/Atmosphere/files/5873829/01611442625_0100000000000014.log)

### System Firmware Version

Firmware: 11.0.1 (Atmosphère 0.17.1-master-49f07b06)

### Environment?

- Hekate Bootloader from which i load Fusee
- Clean install from Github With Meso
- SysModules Running in Background SaltyNX / Tesla /Emuiibo  (also tried to deactivate them all, didnt work either)
- Homebrew software installed: [ ftpo pro /PPSSPP /Retroarcj / SimpleModManager/ sys clk manager (no config set for spyro) /
                                                     TriPlayer / 90DNS Testing Utility / Checkpoint /Daybreak /Edizon / Hekate Toolbox / 
                                                     Homebrew Details / JKSV / ldn_mitm / NX Activity log / NX-Shell / NX Update Checker / 
                                                     NXDumptTool / NX-Themes Installer / vgEdit ]

### Additional context?

found out, that "0100000000000014." seems to be some audio related stuff. Thought, that some sysmodules may take up to much space for the audio buffer or something, but seems not to be the problem. also some games working, some not is also very sus. Could it be Mesosphere ? I'm not that of a Pro into the Homebrew meta and seriously out of ideas (and YES i tried to use google....)

SciresM, 2021-01-26:
"Deactivate" not a real thing.

Does the issue occur with no custom sysmodules whatsoever? Please post a screenshot of /atmosphere/contents to confirm what sysmodules you have.

SciresM, 2021-01-26:
Piracy is not supported.

closed 2021-01-26 by SciresM