Atmosphere-NX/Atmosphere #1349: Not booting after downgrade atmosphere daybreak

lichtlichtlicht, 2021-01-30:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Describe the issue in detail and what you were doing beforehand. - i have nintendo switch lite sxmodchip sx 3.05 fw 10.0.2 the moment atmosphere release support for mariko i already try atmosphere cfw. It boot successfull. But theres a bug when you enter sleep mode it will not wake up so i force restart to work again. I decided to upgrade my firmware using daybreak to 11.01 it boot successful but the bug still persist when you enter sleep mode it will not wake up so you need to force restart to work the switch again. So this is the part the error or big i encounter i downgrade my fw using daybreak to 10.02 and remove atmosphere but when i flash fw 10.02 im trying to turn on but it will not boot there is a error. Fatal error: [NXBOOT] Failed to determine mariko warmboot firware. Please power to reboot </br>
 Did you make any changes related to Atmosphère itself? - no </br>
 If so, make sure to include details relating to what exactly you changed.  -only downgrade</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

10.0.2 update to 11.0.1 downgrade to 10.0.2</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: official
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * no only daybreak]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]
- ## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Describe the issue in detail and what you were doing beforehand. - i have nintendo switch lite sxmodchip sx 3.05 fw 10.0.2 the moment atmosphere release support for mariko i already try atmosphere cfw. It boot successfull. But theres a bug when you enter sleep mode it will not wake up so i force restart to work again. I decided to upgrade my firmware using daybreak to 11.01 it boot successful but the bug still persist when you enter sleep mode it will not wake up so you need to force restart to work the switch again. So this is the part the error or big i encounter i downgrade my fw using daybreak to 10.02 and remove atmosphere but when i flash fw 10.02 im trying to turn on but it will not boot there is a error. Fatal error: [NXBOOT] Failed to determine mariko warmboot firware. Please power to reboot </br>
 Did you make any changes related to Atmosphère itself? - no </br>
 If so, make sure to include details relating to what exactly you changed. -only downgrade</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

10.0.2 update to 11.0.1 downgrade to 10.0.2</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: official
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * no only daybreak]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]
 I can boot to payloader like sxos and hekate i tried to boot using hekate but there is error.
Fatal error: [NXBOOT] couldnt parse boot0: sucess. I made a nand backup using sxos bootloader after i restore it still blackscreen or do i need hekate to restore it? Help

SciresM, 2021-01-30:
You appear to have pasted the template multiple times.

Sleep mode issues --> sounds like your boot0 does not contain valid warmboot firmware. Which matches the error message. I would restore your first nand backup.

"Failed to determine mariko warmboot firmware" shows up when you have downgraded without having booted the upgraded version with atmosphere. Your fuses match 11.x, so you can't downgrade to 10.x without booting 11.x atmosphere first. 

Make sure you're booting via spacecraft etc, your description doesn't make very much sense because you said you booted 10.x/11.x, but atmosphere refuses to boot unless it finds valid warmboot firmware (as you can see from the error post downgrade).

I would also delete the sd:/warmboot_mariko folder, in case you have an incorrect cache or whatever.

lichtlichtlicht, 2021-01-30:
Thanks for the reply. So you mean i will downgrade my atmosphere to boot successfully? I success to boot to 11.0.1 when the sleep mode and not waking up without force restart so i decided to downgrade when im booting it again it shows failed to determine mariko warmboot firmware. How about couldnt parse boot0: success im trying to launch fusee primary to hekate

lichtlichtlicht, 2021-01-30:
Update. Flash the spacenx then install atmosphere but still the error persist. "Failed to determine mariko warmboot firmware". When i try to boot fusee-primary. Bin payload in hekate it says. Fatal error couldnt parse boot0: success

SciresM, 2021-01-31:
You need to have booted the correct version at least once before you can ever boot a downgraded version.

Contact me on discord, SciresM#0524.

lichtlichtlicht, 2021-01-31:
Already sent a request in discord. Name PeakyBlinders

closed 2021-02-04 by lichtlichtlicht