Atmosphere-NX/Atmosphere #1352: Fatal panic when using dns.mitm with youtube ads host blocker

JrRadiant, 2021-02-03:
## Bug Report

</br>
</br>

### What's the issue you encountered?

I edited the /atmosphere/hosts/default.txt with the https://ewpratten.retrylife.ca/youtube_ad_blocklist/hosts.ipv4.txt 

and after I launched the Youtube app, panic occured.

Title ID: 010041544D530000
Error: DABRT (0x101)
fatal-errors/024ca3556</br>

### How can the issue be reproduced?

[ * ]</br>
[ Add the google's ads servers into the  /atmosphere/hosts/default.txt, reboot, and start the Youtube application ]</br>

### Crash Report
[report_024ca9556.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5916913/report_024ca9556.zip)

### System Firmware Version

11.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate chainloading via fusee-primary
- Official release atmosphere-0.18.0-master-26d8db74
- Do you have additional kips or sysmodules you're loading: No.[ * ]

### Additional context?

- Additional info about your environment: No, just vanilla.

SciresM, 2021-02-03:
Could you please enable debug logging (`atmosphere!enable_dns_mitm_debug_log = u8!0x1`), and send me the generated dns_mitm_debug.log?

It looks on casual inspection like nullptr is being passed for the port argument to GetAddrInfoRequest...

JrRadiant, 2021-02-03:
OK:
[*]Added atmosphere!enable_dns_mitm_debug_log = u8!0x1 into system_settings.ini.
[*]Reboot
[*]The panic occurred when system boots with the error:
**Title ID: 010041544D530000
Error: std::abort (0xFFE)**
[report_07349d25.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5916979/report_07349d25.zip)
[*]dns_mitm_startup.log is here:
[dns_mitm_startup.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5916981/dns_mitm_startup.zip)
[*]But there's no dns_mitm_debug.log.:(

SciresM, 2021-02-03:
Ah, that's not quite what's needed. You shouldn't add that line to system_settings.ini, you should add the line "enable_dns_mitm_debug_log = u8!0x1" (without quotes) to the [atmosphere] section of system_settings.ini.

JrRadiant, 2021-02-03:
Oops, got it.

The generated dns_mitm_debug.log is here: 
[dns_mitm_logs.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5917000/dns_mitm_logs.zip)

SciresM, 2021-02-03:
yep: `[01003a400c3da000]: GetAddrInfoRequestWithOptions(s.ytimg.com, (null))`

Okay...I guess some reverse engineering needs to be done to see what official code returns for port number when nullptr is passed as argument.

It'll make it into 0.18.1, which'll probably release in the next few days.

znxDomain, 2021-02-03:
Per [these](https://man7.org/linux/man-pages/man3/getaddrinfo.3.html) docs:

       service sets the port in each returned address structure.  If
       this argument is a service name (see services(5)), it is
       translated to the corresponding port number.  This argument can
       also be specified as a decimal number, which is simply converted
       to binary.  If service is NULL, then the port number of the
       returned socket addresses will be left uninitialized.

And per my testing, a call to `getaddrinfo(hostname, NULL, &hints, &res)` returns `null` in `addrinfo->ai_addr->sin_port`.

[image]

This works properly for a request not MITM'd.  But does abort per this issue when the response comes from DNS-mitm.

It appears [here](https://github.com/Atmosphere-NX/Atmosphere/blob/4fa585a23fb3349e2beaf2f1b21e1f2c37cc6b0f/stratosphere/ams_mitm/source/dns_mitm/dnsmitm_resolver_impl.cpp#L74) that you don't modify the port, calling `InetHtons` seems appropriate, unless that's where it's crashing due to the `null`.  I'm obviously missing something.

fennectech, 2021-02-03:
im also seeing this issue when trying to use dns_mitm to use a Minecraft server   This is what I've added to the default.txt  `#104.238.130.180 geo.hivebedrock.network
#104.238.130.180 hivebedrock.network
#104.238.130.180 mco.mineplex.com
#104.238.130.180 play.inpvp.net
#104.238.130.180 mco.lbsg.net
#104.238.130.180 mco.cubecraft.net
#104.238.130.180 play.galaxite.net`

Am I doing it right  (ignore the commented out lines   I did that so it wouldn't panic while I figure out whats going on

closed 2021-02-03 by SciresM

fennectech, 2021-02-08:
Im still seeing this on the latest version of atmosphere (downloaded today fro settings page.)

fennectech, 2021-02-08:
https://www.dropbox.com/s/9u4oe3neyqeu57t/report_0a72ffc71.bin?dl=0

https://www.dropbox.com/s/igdl0jevavs1jul/report_0000000061cb9d2b.bin?dl=0

Here are the reports

SciresM, 2021-02-08:
I mean, yeah? I haven't done a release that includes the fix commit yet.

fennectech, 2021-02-08:
Ah.  Ok.  I was under the impression that a silent update had been made.  Sorry to waste your time on that.  Ill make my own build until then.

fennectech, 2021-02-21:
After a week of trial and error.  This issue still exists in latest atmosphere (0.18.1.  ).  Spent days with everything off slowly turning my sysmodules back on with no panics.  Then the moment i turned DNS Mitm back on it panics when opening minecraft.   So. This might need to be reopened.  Or should I create a new issue for it.

SciresM, 2021-02-21:
Upload the fatal error report.

fennectech, 2021-02-21:
https://transfer.sh/gHtJP/report_27f1c485a.bin    Here it is

SciresM, 2021-02-21:
This is completely unrelated to this issue, this is a sm fatal, not an ams_mitm fatal.

fennectech, 2021-02-21:
I only get it when DNS_MITM is enabled.  If you need more from me jsut let me know.  Should i create a new issue and put it there?

commit c0e3cee6 (2021-02-03): dns.mitm: allow nullptr for port string (closes #1352)