Atmosphere-NX/Atmosphere #1353: Ask for unaffiliated hid mitm sysmodule

impeeza, 2021-02-03:
Hi I constantly use the homebrew https://gbatemp.net/threads/the-legend-of-zelda-mystery-of-solarus-dx-solarus-engine.500222/, it works fine but never was updated to latest libraries and still relies on old hid-mitm, I did try to compile using news libraries but I have no the knoledge.

Please if possible release a hid mitm sysmodule for supoort these HomeBrew.

thanks a lot.

Masamune3210, 2021-02-04:
one could probably assume that unaffiliated would mean someone not them or official

Slluxx, 2021-02-04:
@impeeza dont update atmosphere if you want to continue playing that game. Or ask in gbatemp for help to recompile it on your own.

SciresM, 2021-02-05:
This doesn't need an atmosphere issue. I'm following the situation.

closed 2021-02-05 by SciresM

impeeza, 2021-02-05:
Thanks, regards from Colombia