Atmosphere-NX/Atmosphere #1354: Album

tsukina22, 2021-02-03:
Hello, i can't get out of the album, after the update, i try to see the album clicking R more Album and it worked, but now that i want to see the Hbmenu doesn't show anything anymore on album, and I'm not clicking R, if I click R together with a game the Hbmenu show upon, but not in Applemode and to play the game is just press R again and the Hbmenu goes out, but in album is not working anymore, help, please.

SciresM, 2021-02-03:
1) you did not use the issue templates
2) I don't even understand your issue.

closed 2021-02-03 by SciresM

tsukina22, 2021-02-03:
> 1. you did not use the issue templates
> 2. I don't even understand your issue.

I don't know what is issue templates, and just because you don't understand my question doesn't mean that nobody will do, i ask with a lot of details. 
My issue is, my hbmenu is not showing up anymore on Appletmod when I click on Album, i already try turn the switch off, but hbmenu doesn't come back to álbum. 
Now can only access the hbmenu clicking on a game with the R pressing. 
I want the Hbmenu to return to album! That is my problem, and i explained everything with a lot of details, is not hard to understand, i just want some help.