Atmosphere-NX/Atmosphere #1356: Blackscreen after Nintendo logo after updated to AMS 0.18.0-master-26d8db74

dezem, 2021-02-03:
## Bug Report

### What's the issue you encountered?
Updated AMS Vanilla 0.18.0
Erista Switch with trinket m0 stays almost in blackscreen after Nintendo logo.
Sometimes it boots, but almost it stays at blackscreen with only background light.
No input possible, must hard power off the switch and try again.
Used emuMMC as hidden partition created by hekate.

### How can the issue be reproduced?
If i revert to atmosphere-0.17.1-master-49f07b06+hbl-2.4.0+hbmenu-3.4.0, it normal boots every time.</br>

### System Firmware Version
FW 11.0.1

### Environment?
- Use the fusèe payload with the trinket m0
- Official release atmosphere-0.18.0-master-26d8db74+hbl-2.4.0+hbmenu-3.4.0.zip
- Do you have additional kips or sysmodules you're loading: Only AMS vanilla and hekate nothing else.
- Homebrew software installed: currently only daybreak
- SD fat32 formatted

payload.bin = fusee-primary.bin
payload1.bin = hekate.bin

[image]

MurdenBtw, 2021-02-04:
I have the same issue except  no version of Atmosphere works for me. It stays on a black screen after injecting the payload and it's been doing this for months now. I Format my SD card to fat32 multiple times and it never worked. switch isn't patched either.

dezem, 2021-02-06:
Solved the problem by myself.

closed 2021-02-06 by dezem

masagrator, 2021-02-06:
>Ask for help because you can't find solution.
>Solve problem.
>Don't write solution.
>New person meets the same problem
>Go to first point

dezem, 2021-02-06:
I don't cleaned the contents folder successfull...this was my fault.
I thought I deleted everything there. 
Sys-ftp-light was still there...this causing my problem.

PixyGixy, 2021-02-08:
> Solved the problem by myself.

bro please i need to know how you did it i also have the same problem

masagrator, 2021-02-08:
> > Solved the problem by myself.
> 
> bro please i need to know how you did it i also have the same problem

...

> I don't cleaned the contents folder successfull...this was my fault.
> I thought I deleted everything there.
> Sys-ftp-light was still there...this causing my problem.

PixyGixy, 2021-02-08:
> I don't cleaned the contents folder successfull...this was my fault.
> I thought I deleted everything there.
> Sys-ftp-light was still there...this causing my problem.

uuuhhhh this made no sense but im guessing you mean wiping everything out of your contents folders work but when i did that its still a blackscreen

Masamune3210, 2021-02-09:
Empty sd card of everything other than Nintendo, bootloader, and switch folders. Extract atmosphere release to root of sd card, attempt to boot

That is what people mean by starting from a clean install