Atmosphere-NX/Atmosphere #1357: Sleep function not working on a specific Lite console

sthetix, 2021-02-04:
## Bug Report

Sleep function not working on a specific lite console.

### What's the issue you encountered?

I got two switch lite, one with a working sleep function and one that does not
both running atmosphere 0.18 on firmware 11.0.1
the sleep function worked well if I launch Atmosphere using Hekate, but I got a black screen when booting Atmosphere directly without Hekate.
And it doesn't matter if we use the spacecraft-nx or the sx gear. Both firmwares have the same issue
the console can launch ofw or stock firmware normally - I even pulled the chip connector to test it, and it can run stock without a problem, and the sleep function works, so I guess the boot0 is no damage.
I tried to regenerate the nand with emmchaccgen and flash it, and it doesn't make any change. The issue persists.
Using daybreak to reinstall the firmware and the same thing, the issue is still there.

### How can the issue be reproduced?

Boot to Atmosphere directly using the spacecraft-nx or the sx gear, without using Hekate. And for a specific lite console, the sleep function is broken. As soon as we press the power button, the console goes to sleep, and we cannot wake it unless we press and hold the power button for 12 seconds to reboot it.

### Crash Report

None generated by Atmosphere

### System Firmware Version

11.0.1</br>

### Environment?

The sleep function doesn't work properly when booting Atmosphere directly, using fusee
But worked well when using Hekate
I am using the Official release 0.18.0
No additional kips
no homebrew apps

### Additional context?

- using a clean and known good micro sd card, SanDisk / Lexar, formatted as fat32. 
- no games or apps installed on the device, pure stock condition

lichtlichtlicht, 2021-02-04:
Same with my switch lite 11.0.1 latest atmosphere 0.18. no games or etc. Only hekate and atmosphere install in sdcard when i enter sleep mode it doest wake i need to force restart or do not sleep my switch to work again

sthetix, 2021-02-04:
> Same with my switch lite 11.0.1 latest atmosphere 0.18. no games or etc. Only hekate and atmosphere install in sdcard when i enter sleep mode it doest wake i need to force restart or do not sleep my switch to work again

Then I guess, it is a software bug. thx for the info.

SciresM, 2021-02-04:
"software bug" no, that guy has a bad SD card. I spent a lot of time debugging with him on discord.

sthetix, 2021-02-04:
oh..I see..then i guess the issue on my lite console is not related to bad sd card, i have tested several sd card from several vendors and capacity and still it wont wake up ..~ unless using Hekate (via launch or via payload menu).

SciresM, 2021-02-04:
@sthetix contact me on discord tomorrow, SciresM#0524.

I don't have the mental energy to deal with more warmboot_mariko sd card nonsense today, but we'll sort out/do debugging builds for you tomorrow as needed.

lichtlichtlicht, 2021-02-04:
I already change my sdcard to another fat32 128gb put my game at least 7 so i can sure my sdcard is not a problem because this is new sd

SciresM, 2021-02-05:
Closing, in favor of making a new issue.

closed 2021-02-05 by SciresM