Atmosphere-NX/Atmosphere #1358: Panic when trying to run sys-modules (emuiibo, sys-clk)

jacoghi, 2021-02-04:
## Bug Report

When trying to toggle modules using the tesla menu, sys-clk, emuiibo, and others, Atmosphere goes into a Panic event, Title ID 01000000000000004 (might have more or less zeros...)
Error: std:: abort (0xFFE)
Logs generated, Logs attached

### What's the issue you encountered?

Toggling modules on/off, going into the emuiibo tesla overlay menu.

### How can the issue be reproduced?

Clean system, going into the tesla menu.

### Crash Report

See attached here
https://drive.google.com/file/d/1s5O3eskrIg7dvG9ZbPZoVGcHPutuYiGn/view?usp=sharing

### System Firmware Version

11.0.1

### Environment?

fusee-primary.bin
Modules installed, all latest: emuiibo, sys-clk, system-monitor, ldn_mitm (not used)

### Additional context?

masagrator, 2021-02-04:
System monitor is not a module.

SciresM, 2021-02-04:
Atmosphere can probably handle this more gracefully than by aborting, but those sysmodules need to be updated by their developers -- they're relying on the sm extension that was deprecated two months ago, and removed in 0.18.0.

closed 2021-02-04 by SciresM