Atmosphere-NX/Atmosphere #1359: [ Known ] Reports of wake/sleep failure when booting fusee-without-chainload-from-hekate on some hoag units.

SciresM, 2021-02-05:
Presumably something missing in hardware init....

...this may just not be addressed until fusee rewrite, because fusee is pretty unmaintainable?

fille1976, 2021-05-02:
I have also sleep problem with switch lite,im using sxlite modchip with spacecraftnx flashed succesfully.
I use atmosphere 0.19.2 and im on firmware 12.0.1,all working fine,except the sleep mode,it goes in sleep,but you can see the screen isnt off,and when try to get out of sleep mode,it doesnt start,have to hard reset.
I use now sandisk 128gb formatted as fat32.

Moriarty92x, 2021-05-15:
I'm having the same problem as Fille. Updated Atmosphere, Hekate and sigpatches to latest versions. Updated emunNAND to 12.0.1 with daybreak. Reset, everything worked fine. Later I updated sysNAND to 12.0.2 and now I keep getting this error: "Failed to match warm boot with fuses - Sleep Mode won't work".

GavinDarkglider, 2021-05-17:
Well, that sounds like nintendo is burning fuses with updates, and the warmboot stuff needs a new dump of the fuses to work properly with this update. That is just a guess though. Also, dont post about sigpatches, it is well known that Atmosphere devs cant and wont support issues if sigpatches are used, due to the legality issues that could arise, so dont mention them later.

closed 2021-10-16 by SciresM