Atmosphere-NX/Atmosphere #1362: booting atmosphere shows with a black screen

PixyGixy, 2021-02-07:
## Bug Report

### What's the issue you encountered?
I think I downloaded some kind of overlay for tesla and something else through ftpd and then a couple hours later atmosphere crashed so I just decided to reboot and when I got into my home menu after 10 seconds atmosphere decided to crash again and doing the exact same again when I tried rebooting again so then I decided to switch my version of atmosphere to the latest and copied all the old folders like contents, titles and config to the latest version and now when I try booting atmosphere it shows the loading screens halfway and then just goes black after the atmosphere logo with black backround screen. 

### How can the issue be reproduced?

above this question

### Crash Report

I've went inside my crash reports and I only saw previous crashes and not the one where it was loop crashing every time I reboot atmosphere 

### System Firmware Version

uh i think it was 11.0.0

### Environment?

fusse-primary

- Do you have additional kips or sysmodules you're loading: emuibo,mission control, triplayer
- Homebrew software installed: yeah uh plenty ill list them if somebody asks 

### Additional context?

uh i use custom backrounds?

SciresM, 2021-02-08:
Does the error occur without custom sysmodules/etc?

PixyGixy, 2021-02-08:
yeah still does

SciresM, 2021-02-08:
What are the crash details? Send a screenshot of the crash.

SciresM, 2021-02-08:
Ah, black screen, not crash.

Send a screenshot of `/atmosphere/contents`?

PixyGixy, 2021-02-08:
[image]
[image]

SciresM, 2021-02-08:
"Does the error occur without custom sysmodules"

You have 8 custom sysmodules, if not more, my count.

Delete all the custom sysmodules, clean extract latest atmosphere release zip. 

It will fix your problem, with very high certainty.

PixyGixy, 2021-02-08:
I really don't know much about sysmodules I thought you were talking about tesla menu, can you specify which of these folders is a sysmodule. also "clean extract latest atmosphere release zip." do you mean like delete everything in atmosphere and extract a clean version of atmosphere but then I think some homebrew nros will stop working well im not sure just correct me on this

josegonzalez, 2021-02-08:
I'm seeing the same issue. Attempting to send the latest `fusee-primary.bin` works "fine" from the `fusee-launcher.py`, but I get a black screen. I'm using the following release:

```
https://github.com/Atmosphere-NX/Atmosphere/releases/download/0.18.0/atmosphere-0.18.0-master-26d8db74+hbl-2.4.0+hbmenu-3.4.0.zip
```

Will attempt to clean out atmosphere etc. and try again.

josegonzalez, 2021-02-08:
I cleared out everything and retried with the same install, seems to have fixed that issue, though it then started not loading custom NSPs (yes, I realize thats not in scope for this project).

In any case, a clean extraction appears to allow atmosphere to load again.

SciresM, 2021-02-08:
Piracy is not supported.

closed 2021-02-08 by SciresM

josegonzalez, 2021-02-09:
I don’t think anyone here is trying to get support for piracy. This is strictly whether the screen loads at all after an upgrade “over” an existing install.

Masamune3210, 2021-02-09:
You mentioned custom NSP files not working, that is piracy and is not supported even by association

josegonzalez, 2021-02-09:
I was referring to [this thing](https://gbatemp.net/threads/nro2nsp-retroarch-rom-nro-forwarders.519915/) that I’m using to create these weird forwarder nsp  for retro arch and that homebrew Zelda game, didn’t realize that was piracy (I guess it must be if it injects something onto the home screen? Idk).

I ended up downgrading so the home screen versions would load again.