Atmosphere-NX/Atmosphere #1364: Game Freezing on Text/Controller Box Open

JackTYM, 2021-02-08:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br> When playing a game (In this example, BOTW) and the system tries to open the text box or (in another case, Mario Kart 8) tries to open the controller connection screen, the game will freeze and no more inputs can be done to that game unless you close it and reopen it.
[ Did you make any changes related to Atmosphère itself? ]</br> I have some other cfw apps installed, though i don’t think those are the problem, if needed I can list them though
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br> Open any game that requires using the text box to name something or a local multiplayer game that requires selecting the controllers. When it is supposed to open the screen for said action, it will freeze.

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

11.0.1</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ] Hekate launching Fusèe-Primary
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ] Awoo Installer, btdb, Cheats Updater, Checkpoint, Daybreak, EdiZon SE, Fizeau, GameCard Installer, Goldleaf, HB App Store, Homebrew Details, Lockpick, MM-LINX, Mod Manager, NX Activity Log, NX-Shell, nxsh, nxdumptool, NXThemesInstaller, pplay, RCMReboot, RCMRebootInstant, RetroArch, sys-clk manager, SysDVR Settings, Tinfoil Installer, TriPlayer, vgedit, 

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2021-02-08:
I am almost entirely sure the issue will be some custom sysmodule you're running.

What custom sysmodules are you running?

Does the error occur without them?

JackTYM, 2021-02-08:
> I am almost entirely sure the issue will be some custom sysmodule you're running.
> 
> What custom sysmodules are you running?
> 
> Does the error occur without them?

sorry, sort of new to the cfw / modding community, what is a sysmodule?

SciresM, 2021-02-08:
Extension components to the cfw/operating system you have running in the background. Closing this issue because it'll be sysmodules...seek support in the Nintendo homebrew discord, or just delete your /atmosphere/contents folder and extract clean release zip without adding custom modules.

closed 2021-02-08 by SciresM