Atmosphere-NX/Atmosphere #1367: [ERROR] Fatal error! An exeception occured! Exception type: Data abort (please see image)

akirayuki, 2021-02-10:
## Bug Report

### What's the issue you encountered?

Upon injecting fusee-primary.bin, I get this
[image]

### How can the issue be reproduced?

Injecting the payload in the switch again.

### Crash Report

No crash report

### System Firmware Version

11.0.1

### Environment?

vanilla fusee-primary.bin

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Mistera91, 2021-02-10:
Same. I think it's not the SD card cause I reformatted it and still the same error. Do you tried an older version of atmosphère ? I'm gonna try this now.

Mistera91, 2021-02-10:
Ok, idk how I did that but i just re-refreshed my sd files with ams 0.16.0 and it works

Mistera91, 2021-02-10:
Only problem is now I have nothing, no mods, no screens, no games.

Masamune3210, 2021-02-11:
homebrew can be copied back to the sd card by just re-copying it to the switch folder, creating it if it doesn't exist on the root of the sd card. As for the games being missing, piracy isn't supported. If you own them legit, just redownload them from the eshop

jacoghi, 2021-02-18:
You're probably injecting an older version of fusee with a newer build of Atmosphere. Download 0.18.1 and its respective fusee and try again.

Judgegeo, 2021-02-19:
Has this been fixed? I'm getting the same error with the latest JITS release and also this base release.

SciresM, 2021-02-19:
Pretty sure this is just because fusee is currently broken on exactly 6.2.0, because the mariko refactor broke tsec emulation.

yoohahn, 2021-02-22:
Have the exact same issue. Had to downgrade to 0.16.2 to get it to work.

SciresM, 2021-09-07:
fusee-primary no longer exists, so I'm closing this issue, which was solved by 0.20's tsec keygen change anyway.

closed 2021-09-07 by SciresM