Atmosphere-NX/Atmosphere #1369: [Error] 90DNS not working with 0.18

ZaydelSenpai, 2021-02-13:
Hello, i decided to update ATM from 0.16.2 to 0.18 (Using 90DNS), i was just curious about the new DNS block system and when i use the 90DNS Checker i got unblocked on all tries, also without the "Hosts Folder" it's the same i deleted my wifi and added it again and i still getting Unblocked for all the entries.

SciresM, 2021-02-13:
1) You did not use the issue templates.

2) The default DNS.mitm redirections are not the same as 90DNS -- if you want an equivalent to 90DNS, you'll need to create a custom hosts file. Seek help on the Nintendo homebrew or ReSwitched discords for this, if needed.

closed 2021-02-13 by SciresM

shadow2560, 2021-02-16:
I use this issue to ask a little question on DNS hosts file redirection (sory but I'm not realy familiar with Discord), if I have this example:

```
127.0.0.1 *.foo.net

1.2.3.4 test.foo.net
```

The second rule tataly cancel the first rule or just add the exception of "test.foo.net" to the redirection for "*.foo.net"?

SciresM, 2021-02-16:
The last line in the file which matches the domain is used.

So it's like an exception.

test.foo.net will go to 1.2.3.4, all other *.foo.net will go to 127.0.0.1.

shadow2560, 2021-02-16:
That's what I was thinking but I prefered a confirmation, thanks a lot for the very quick response.