Atmosphere-NX/Atmosphere #1371: Breath of the wild layeredfs mods cause dlc to be undetected by the game

XAEA-XII2, 2021-02-15:
## Bug Report

### What's the issue you encountered?
When using more than 2 layeredfs mods, breath of the wild stops detecting DLC.
I tested around 12 different mods and all of them had the same result on 0.18.0.
I tested a fresh install of 0.17.1 and the issue didn't occur. The mods worked properly.
I tested a fresh install of 0.18.0 and the issue occurred again.

Two other people did the same test and reported the same issue.

### How can the issue be reproduced?
Have more than 2 layeredfs mods for breath of the wild and DLC won't be detected anymore by the game.

### Crash Report
N/A
### System Firmware Version

11.0.1</br>

### Environment?
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  - Hekate
- Official release or unofficial build: 
  - Official release version 0.18.0
- Do you have additional kips or sysmodules you're loading:
  - No
- Homebrew software installed:
  - N/A

### Additional context?

- Additional info about your environment:
N/A

NicholeMattera, 2021-02-15:
I'm using Atmosphere 0.18.0 on firmware 11.0.1 with "Zelda's Ballad & More" and "No slipping while climbing in the rain" mods with no issues accessing the DLC.

SciresM, 2021-02-15:
This doesn't really make any sense to me, given how dlc detection works...can you upload the romfs_metadata.bin generated when you launch the game with multiple mods, and share a screenshot of your romfs folder?

XAEA-XII2, 2021-02-15:
> This doesn't really make any sense to me, given how dlc detection works...can you upload the romfs_metadata.bin generated when you launch the game with multiple mods, and share a screenshot of your romfs folder?

I currently have 3 mods installed (I did test this with a few others, and I had the same issue):

- Infinite Durability mod

- Better Link House

- Star Wars Speeder Bike

Here is the `romfs_metadata.bin` file and here is a screenshot of the romfs folder (there is only one folder inside).
[image]

[romfs_metadata.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5982702/romfs_metadata.zip)

SciresM, 2021-02-15:
It looks like the built romfs doesn't contain AocVersion.txt. Bizarre.

I'll look at this later today.

SciresM, 2021-02-16:
reproduced this locally.

commit 74e4e700 (2021-02-16): fs.mitm: fix cache of non-current-process data storages (closes #1371)
closed 2021-02-16 by SciresM