Atmosphere-NX/Atmosphere #1372: Why not using dedicated partition to CFW ?

GGLinnk, 2021-02-15:
## Feature Request

### What feature are you suggesting?
#### Overview:
I was wondering, why not use a dedicated partition for storing the main components of the custom firmware.

#### Smaller Details: 
- GPT will be needed if GUID is used.
- If GPT can't be used we can use files 

#### Nature of Request:  
- Change the way that user/dev can manage and update CFWs.

### What component do you feel this would best fit within?
- [Fusée](https://github.com/Atmosphere-NX/Atmosphere#components)
  - Atmosphere's custom bootloader ?

I don't really know...

### Why would this feature be useful?
[image]
By bootloader I mean something like hekate.

I don't know if it's really achievable, but it's an idea I have for a long time in my head...

SciresM, 2021-02-15:
This would prevent updating/installing atmosphere by dragging the contents of a zip to the sd card, which is the standard/intended method.

It would also make development a huge pain in the ass.

Gonna be honest and say I don't actually think this is desirable -- the bootloader partition thing I'm against (even ignoring the fact that "universal bootloader" is easier said than done), and I definitely object to explicit recognition/support of those clone packs that repackage atmosphere and some homebrew with no added value.

SciresM, 2021-02-15:
Also, GPT can't be used. Nintendo only supports MBR on SD.

closed 2021-02-15 by SciresM