Atmosphere-NX/Atmosphere #1373: LEGO Marvel Superheroes 2 boots with a blackscreen

OwenJAYEH, 2021-02-16:
## Bug Report

### What's the issue you encountered?

When launching LEGO Marvel Superheroes 2 in Atmosphere sysCFW with fusee-secondary.bin, the game will get stuck on a black screen following the Nintendo Switch splash screen. When launching with stock sysNAND, it will instantly load. The only things I reckoned would have prevented it from loading were EdiZon / sys-clk, but even when uninstalling both, the game still hung up on a black screen. It seems to be the only game in my library that does this.

Additionally, I am able to press the home menu and be back on my Switch home menu just fine. I can also open up Tesla overlay just fine. Neither Atmosphere, the Switch, or the game crashes.

### How can the issue be reproduced?

1. Bootload into Atmosphere
2. Launch LEGO Marvel Superheroes 2

### Crash Report

N/A

### System Firmware Version

11.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  hekate with fusee-secondary.bin
- Official release or unofficial build: 
  hekate v5.5.2 and Atmosphere 0.17.0
- Do you have additional kips or sysmodules you're loading:
  I do not
- Homebrew software installed:
  90DNS Testing Utility, Atmosphere-Updater, Cheats Updater, Checkpoint, Daybreak, EdiZon, ftpd snap, HB App Store, NX-Shell, NXThemes Installer, sys-clk manager, and Ultimate Mod Manager

### Additional context?

Like I said, this is the only game this happens to. It almost seems like the game has an incompatibility with Atmosphere, which I know is completely unheard of.

I've also tried
- booting into emuMMC with fusee-secondary.bin
- Disabling and uninstalling both EdiZon and sys-clk
- Checking LEGO Marvel Superheroes 2 for corrupt data
- Reinstalling LEGO Marvel Superheroes 2 from the eShop

SciresM, 2021-02-16:
Have you tested with truly clean/stock atmosphere? Send a screenshot of the /contents/ folder of the clean atmosphere you tested with?

OwenJAYEH, 2021-02-16:
Heh, well I feel a little bit silly. Trying it with a stock version of Atmosphere worked perfectly. Thank you.

closed 2021-02-16 by OwenJAYEH