Atmosphere-NX/Atmosphere #1374: Tesla don't work

sanpietro2000, 2021-02-16:
Bug Report

Atmosphere 0.18.0 don't work with tesla.
Atmosphere don't boot cfw with error code 2021-0003 (0×615)

System Firmware Version

11.0.1

 Environment

Hekate with last pack 5.5.4 and atmosphere 0.18.0

rashevskyv, 2021-02-19:
Works fine for me. Maybe you should make issue on tesla's git for tesla error?

sanpietro2000, 2021-02-19:
No, i have change sx os mariko because tx now don't upgrade anything. I
have downloaded atmosphere and hekate at least version. After do this i
have downloaded edizon at least version and tesla but atmosphere don't
start. I've eliminated overlays folder on switch and tesla contents from
atmosphere and all now is working again. But tesla don't wont to work

rashevskyv, 2021-02-19:
I have tried on both erista and mariko with latest atmosphere and tesla. Boot with fss0. On both working without issue. 
Try another card and update to the latest atmo first. You always **must** be on the latest atmosphere before reporting a problem.

sanpietro2000, 2021-02-19:
I've tried also with atmosphere 0.18.1 and with another sd card because i
want to do this thing by myself. But this time i don't really know what can
i have to do. If you want i can do a video for you

javastate, 2021-02-21:
I have having the exact same issue with Atmosphere 0.18.1 and Tesla Menu. I installed tesla on a fresh installed Atmosphere 0.18.1. My atmosphere works perfectly fine after fresh install, but as soon as I installed the tesla, it won't be able to boot into atmosphere cfw. I crashes right after showing the Nintendo logo. However, as soon as I removed the tesla folder(atmosphere/contents/420000000007E51A), I can boot into atmosphere cfw just fine.

I followed this tutorial step by step to install the atmosphere:
https://nh-server.github.io/switch-guide/user_guide/emummc/sd_preparation/ 

I used TegraExplorer.bin to format my sdcard, hekate_ctcaer_5.5.4.bin as the payroad. I did this with my Mac computer.

Any hint to solve this issue would be greatly appreciated

javastate, 2021-02-22:
I found the solution here.
https://github.com/WerWolv/Tesla-Menu/issues/27

Fixing the archive bit was the solution.

closed 2021-02-23 by SciresM