Atmosphere-NX/Atmosphere #1378: A fatal error while loading atmosphere ID:010041544d530000

Zira0711, 2021-02-23:
## Bug Report

### What's the issue you encountered?

Today I updated switch from 9.0.1 to 11.01(exFat since my SD card is exFat) and also the newest files from sdsetup.com. I did delete all related files and folders before moving the new ones. However, the fatal error occurs after trying to load CFW.  I searched for similar issues online and tried Fix Archive Bit but it is not fixing the issue.

A fatal error occurred when running Atmosphére.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

### How can the issue be reproduced?

Boot to RCM
Launch Hekate
Boot into CFW

### Crash Report

[report_0000000007800e4b.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6028430/report_0000000007800e4b.zip)

### System Firmware Version

11.0.1

### Environment?

hekate 5.5.4

### Additional context?

- Windows 10 is always trying to format SD card (exFat) after taking out from the switch. It complains Corruption was found while examining the boot region. I use chkdsk /F to fix the issue then it works. But I have to redo it  when the fatal error occurs in the switch.

masagrator, 2021-02-23:
Then don't use exfat as it was 1000x said that it's corrupting sdcard.

Zira0711, 2021-02-23:
yes, I understand fat32 should be the obvious choice right now, but back then there is an argument for exfat and I have used this card for 2-3 years. If the issue is actually caused by the sd card I will try to reformat it. Thanks.

SciresM, 2021-02-23:
The error is fs::ResultInvalidFatFormatSd.

So, yeah, reformat your SD card. It's corrupted.

closed 2021-02-23 by SciresM

Hakkai94, 2022-12-18:
Hello. I have this error, so formated my SD card in FAT32 mode, copied Atmosphere 1.4.0 Final on it then started my Switch : same error :(

ElementalTJ, 2022-12-20:
Hello,
I have recently been having this same error.
I have replaced SD Card Reader, reformatted SD card, tried doing a bit-fix, checking SD card with h2testw (passes).
I am almost ready to send my device in for repair.
Does anyone have a fix? Or is my Switch basically dead?

Frequently comes up with 'SD Card Has Been Removed' error as well.
Thanks to all that can help!

masagrator, 2022-12-20:
> Hello, I have recently been having this same error. I have replaced SD Card Reader, reformatted SD card, tried doing a bit-fix, checking SD card with h2testw (passes). I am almost ready to send my [...]

Hardware issues are unrelated to Atmosphere. Seek help in other sources like ReSwitched discord. This does sound like PCB socket connecting your sdcard reader is damaged. Go to repair service

arkilis, 2023-01-03:
same error with firmware 15.0.1. SD card in FAT32 format.

Hakkai94, 2023-01-04:
I found this solution that worked for me :
In Atmosphere, go to "tools" > "Arch bit . RCM . Touch . Pkg1/2" > "Fix
Archive Bit"

Jrmewam, 2023-11-11:
The only working solution for me was to delete the atmosphère folder and replace it by a brand new fresh one. 
After that I was able to copy my content folder back to it.
Solution given by wonderful person on ARG guilded server

Masamune3210, 2023-11-11:
Guys, this issue is closed. Please either go somewhere else for support or open a new one if you have a proper issue to report.

xuanye, 2024-09-21:
> I found this solution that worked for me : In Atmosphere, go to "tools" > "Arch bit . RCM . Touch . Pkg1/2" > "Fix Archive Bit"
Thanks a lot , it worked for me