Atmosphere-NX/Atmosphere #1384: Enable running game card backups without needing to convert and install them.

eyeonus, 2021-02-25:
## Feature Request
### What feature are you suggesting?
#### Overview:
- The ability to mount backups of my game cards without needing to install them.</br>

#### Smaller Details: 
- N/A

#### Nature of Request:  
- Addition

### What component do you feel this would best fit within?
Unknown, probably one of the following.
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

### Why would this feature be useful?
- This is an alternative to the currently only available option of installing said backups to either my emuNAND or sd card, neither of which are anywhere nearly large enough to store all of my games. This requires me to either carry around the game cards for all the games I don't have currently installed if I want to play them (which also means removing whichever one is in the game slot every time I want to play a different one of these not-installed games), or to delete ones that are installed in order to make room for ones that are not, neither of which is conducive to my desired goal of having all of my games available to me without having to have a case full of game cards / sd cards / whatever that need to be swapped out whenever what I want to play isn't on the media currently inserted into the Switch. As it is right now, I have one sd card for me, one for my wife, one for my kid, and one for Skyrim- _just_ Skyrim, and it's getting to the point where my kid's going to need a second sd card. It is a hassle having to power off the Switch, swap cards, and reboot, especially since neither of the other two are willing (in the case of wife) or trusted enough (in the case of son) to go through the process.
- We keep the game card in their boxes on the shelf and run the games via backup primarily for ease of use, as being able to switch between games without having to swap physical media in and out of the Switch is extremely convenient, and not having to bring a carrying case full of game cards on long road trips saves us all tons of hassle.
- This would also free up valuable hard drive and sd card space for the important stuff -- saves -- and extend the longevity of both the internal hard drive of the Switch and the SD cards, as it would decrease the number of writes to the drive of both.
- Since game card backups can have updates and DLC integrated into the base file, this also means users can have a single file in their archive for their games, rather than needing to have the dump of their game card, plus another file containing the latest revision, plus one file each for every DLC they purchased, making archiving much cleaner and easier to deal with.

SciresM, 2021-02-25:
Piracy is not supported.

If you want to play the game without the physical card inserted, purchase a digital copy on the eShop.

closed 2021-02-25 by SciresM

eyeonus, 2021-02-25:
What are you talking about? This has nothing to do with piracy. Did you not read?

SciresM, 2021-02-25:
Backups are piracy.

eyeonus, 2021-02-25:
I don't see how backing up, say LoZBotW, and installing it using YOUR tools, is any more piracy than downloading LoZBotW from the eShop, dumping it to my computer using YOUR tools, and installing it again later, again using YOUR tools.

I also don't see how taking that same downloaded from eShop LoZBotW, and converting the downloaded files into a format that can put the base game, the update, and the two dlc into a single file, somehow makes it piracy, and yet somehow the original files that were used to make it are not.

You are being dismissive of a truly useful file format that has significant advantages because of what reason? A FILE FORMAT is somehow automatically piracy?

eyeonus, 2021-02-25:
Oh and by the way, backups aren't piracy. SHARING them is.

SciresM, 2021-02-25:
I don't know why you capitalized YOUR. 

I don't support piracy, not have I made any piracy tools.

To install any "backups" on atmosphere you literally have to download patches that modify atmosphere's code to disable anti-piracy checks.

We do not make these (obviously), and we don't approve of them.

Your take on copyright law is incorrect, but more importantly doesn't reflect the stance of the Atmosphere project.

Have a good one.