Atmosphere-NX/Atmosphere #1386: Atmosphere Crashes on Atmosphere

AwesomeKirito, 2021-02-26:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br> I just went to the album and it crashed. Same goes for holding down r 
[ Did you make any changes related to Atmosphère itself? ]</br> Ive added mods to my games
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>https://gist.github.com/AwesomeKirito/c0ccda46ba163dc8a73fde2ff9aed0c1
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>11.0.1
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee 
- Official release or unofficial build: Official
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: Tesla
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2021-02-26:
Does the issue occur on stock atmosphere without custom symodules?

Please post a screenshot of your /atmosphere/contents folder.

The error is `fs::ResultProgramInfoNotFound`, which is an *extremely* strange error for you to be receiving, unless you have some pretty eldritch shit installed.

AwesomeKirito, 2021-02-26:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

AwesomeKirito, 2021-02-26:
i havent tried it on stock

SciresM, 2021-02-26:
Congrats, you now hold the record for the most custom sysmodules I have ever seen installed, with 11 (unless I'm missing some).

You're probably running into some limit on the number of programs that FS can know about simultaneously. Use fewer custom system modules at once.

closed 2021-02-26 by SciresM

AwesomeKirito, 2021-02-26:
thank you it worked