Atmosphere-NX/Atmosphere #1387: Error at boot just after the Switch screen

lashoun, 2021-02-26:
## Bug Report

### What's the issue you encountered?

Atmosphere displays an error screen after boot. 

### How can the issue be reproduced?

I followed the procedure [here](https://nh-server.github.io/switch-guide/extras/updating/):
- I updated Atmosphere from 15.0 to 18.1 and Hekate from 5.3.4 to 5.5.4
- I updated my sysMMC from firmware 10.2 to 11.1 through the system settings
- I updated my emuMMC from firmware 10.2 to 11.1 with `TegraExplorer.bin`
- after rebooting to Atmosphere, Atmosphere showed an error.

### Crash Report

N/A

### System Firmware Version

sysMMC is in 11.1 (I can still boot to it without issue)
emuMMC is supposed to be upgraded from 10.2 to 11.1 but I haven't been able to check.

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate 5.5.4
- Official release or unofficial build: 
  - Official release18.1
- Do you have additional kips or sysmodules you're loading: I don't think so, I don't even know what a sysmodule is
- Homebrew software installed: EdiZon, Moonlight, SkyNX, Tinfoil, Daybreak

### Additional context?

- Additional info about your environment: N/A
[image]

SciresM, 2021-02-26:
delete or update your custom theme.

if you want to delete it, the folder is at /atmosphere/contents/0100000000001000, delete that.

closed 2021-02-26 by SciresM

lashoun, 2021-02-26:
I confirm it solved the issue. Thanks a lot, I'll keep that in mind next time.