Atmosphere-NX/Atmosphere #1389: [BUG] Nsp Forwarder not working

TheMisterPenguin, 2021-02-27:
## Bug Report
### What's the issue you encountered?
After updating from Atmosphere 17.0 with Mesosphere to Atmosphere 18.1 with Mesosphere no nsp forwarder functions.
I tried reinstalling it but it didn't work
The switch shows "Could not start the software. Please try again from the HOME Menu."

### How can the issue be reproduced?

This issue can be reproduced by opening a nsp frowarder from the HOME menu.</br>

### System Firmware Version

11.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
 Hekate 5.5.2
- Official release or unofficial build: 
  Official release version 18.1.0
- Do you have additional kips or sysmodules you're loading:
 Tesla menu, SaltyNX, emuiibo, sys-ftpd-light
- Homebrew software installed:
 Tinfoil, Checkpoint, Kosmos Toolbox, NX-Shell, Retroarch

### Additional context?

- Additional info about your environment:
Using custom themes and updated because Tinfoil was having issues.

Masamune3210, 2021-02-27:
Piracy is not supported

TheMisterPenguin, 2021-02-27:
I don't understand I don't use piracy I just use tinfoil to install my nsp forwarders and all the homebrew apps were from a pack so I just don't use some of them

fincs, 2021-02-27:
Piracy is not supported. **Homebrew is NRO, never installable nsps** (not to be confused with exefs nsps). Installing installable nsps require modifying Atmosphere to disable checks Nintendo put in place to prevent the installation of pirated software.

closed 2021-02-27 by fincs