Atmosphere-NX/Atmosphere #1390: Feature Request: Override screencap defaults

Slluxx, 2021-02-28:
## Feature Request
### What feature are you suggesting?
#### Overview:
- The possibility to override the ability for screenshots or videos either globally or per title

#### Nature of Request:  
- Addition

### What component do you feel this would best fit within?
- Probably Troposphere but i don't know.

### Why would this feature be useful?
Some games do not allow screenshots and videos to be made. Sharing those would be nice too.

SciresM, 2021-02-28:
Closing this since this bridge will have to be crossed naturally during tma2 development anyway.

closed 2021-02-28 by SciresM

Slluxx, 2021-02-28:
tma2 development? Whats that? And is this on the roadmap?

SciresM, 2021-02-28:
See open PR/commit history this month, and yes it's in the roadmap.