Atmosphere-NX/Atmosphere #1392: Error Kernel Panic 0xf00

clockterra, 2021-03-02:
## Bug Report

A fatal error ocurred when running atmosphere. </br>
Program ID: 524850534f53454d </br>
Error Desc: Kernel Panic (0xf00) </br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

11.0.1</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- bootloader fusèe, fuse-primary for 18.1
- Official release version 11.0.1 
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: only the hbl and hbmenu that comes with atmosphere 18.1

[report_000000000248f48d.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6068495/report_000000000248f48d.zip)

SciresM, 2021-03-02:
`Do you have additional kips or sysmodules you're loading: No`

This seems false.

The crash is an invalid memory access during KIP blz uncompress.

Screenshot of your /atmosphere folder? Screenshot of your /atmosphere/kips folder?

clockterra, 2021-03-02:
ohh true i had fs patches too, i found the problem i tried to delete fspatches but for some reason the folder get corrupted and fs patches were not deleted and that was doing the problem, as soon i done a archive bit and put the fs patches on the folder again it worked.
For that i thing that the problem was on my SD with corruption on exfat, now atmos 18.1 is working fine.

closed 2021-03-02 by clockterra

Kokalight, 2024-05-04:
Hello, I have the problem too. I didn't understand how you did it. can you explain it to me again?