Atmosphere-NX/Atmosphere #1394: LayeredFS will not load DLC when dumped into romsFS folder

Hassas99, 2021-03-04:
### What's the issue you encountered?
The game used for testing was DOAX3 Scarlet. DLC is completely broken when loading romfs mods.
DLCs were dumped using nxdumping tool

### System Firmware Version

Latest release</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  - Official release on the latest build using  fusèe.

### Additional context?

In the patch notes you said you addressed this issue however it is still seems to be broken.

"A bug was fixed that caused ams.mitm to incorrectly cache data storages.
This potentially broke DLC when using romfs mods, and could have caused other issues (e.g. with custom themes, and maybe other cases)."

SciresM, 2021-03-04:
The issue you reference is completely fixed. Anything you're experiencing is necessarily a different issue.

"DLC is completely broken when loading romfs mods. I'm not finding any issues when swapping main game files."

So, what, you're attempting to modify the DLC romfs?

SciresM, 2021-03-04:
"completely broken" in what way? Be more specific about what actions trigger what effects.

Hassas99, 2021-03-04:
> "completely broken" in what way? Be more specific about what actions trigger what effects.

I wasn't modifying anything. DLC is kept intact without any modification.

SciresM, 2021-03-04:
I have no idea what you're trying to do, then, and similarly have no idea what's actually wrong.

Hassas99, 2021-03-04:
> I have no idea what you're trying to do, then, and similarly have no idea what's actually wrong.

I was trying to see if the DLC loads from romfs when dumped using nxdumptool, it doesn't.

SciresM, 2021-03-04:
...why would you expect it to?

SciresM, 2021-03-04:
...piracy is not supported.

I am really not sure why you thought that would work.

closed 2021-03-04 by SciresM