Atmosphere-NX/Atmosphere #1395: fatal error 010000000000002b

dickson125, 2021-03-05:
## Bug Report

### What's the issue you encountered?

A fatal error occurred when running Atmosphere.
Title ID: 010000000000002b
Error Desc: std : : abort ( ) called (0xffe)
Report saved to /atmosphere/fatal_errors/report_000000000930b2cd.bin
Press POWER to reboot.

### Crash Report

[report_000000000930b2cd.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6093376/report_000000000930b2cd.bin.zip)

### Environment?

-  hekate

### Additional context?

-  I forgot the system firmware version
-  I cannot get back/don't know how to get back to hekate menu.

SciresM, 2021-03-05:
1) You are not using atmosphere-0.18.1.
2) You probably need to run the archive bit fixer, given this zip has a __MACOSX folder in it, and you're getting "path not found" errors.
3) Hold the power button for 12 seconds to force a shutdown, then you'll boot into RCM and can inject hekate as usual.

closed 2021-03-05 by SciresM

dickson125, 2021-03-05:
Problem fixed after running the fixer, thank you so much.