Atmosphere-NX/Atmosphere #1396: Fatal error: std::abort() called (0xffe). Program ID: 010000000000bd00

TidusWulf, 2021-03-08:
Report attached. This happened after updating to 11.0.1 using daybreak. I have always updated with Choi in the past. Before updating firmware, I made sure atmosphere was up to date and working. The crash occurs after a few splash screens.
[report_000000000afa023b.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6099345/report_000000000afa023b.zip)

TidusWulf, 2021-03-08:
Nvm. Mission Control needed to be updated. Sorry for the trouble. Hopefully someone with this same problem in the future will google search and see this solution. Thank you.

closed 2021-03-08 by TidusWulf

wangyuanpeng5e, 2024-09-17:
Geez… Thank you very much for sharing that information. My data was successfully saved. Thank you!!!