Atmosphere-NX/Atmosphere #1399: Fatal Error Occurred

EggShweg, 2021-03-10:
## Bug Report
### What's the issue you encountered?

I was having trouble with retroarch so I wiped the folder and put a clean install of it on my SD card.
I also placed an updated theme layout in /switch/themes. Now when I try to open hbmenu I get "Fatal Error Ocurred: Title ID: 010041544D530000

### How can the issue be reproduced?

Clean install of RA
update a home themed layout
boot into fusee
open hbmenu
explode?

### Crash Report

I have a fatal error bin linked below.
### System Firmware Version

11.0.1

### Environment?

Hekate > Payloads > fusee
- Do you have additional kips or sysmodules you're loading: just additional sys-clk GUI from the repo
- Homebrew software installed: [ * ]
[report_021d7f3aa.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6118903/report_021d7f3aa.zip)

SciresM, 2021-03-10:
invalid session when doing mitm service -> clone.

I think you're running outdated system modules, or too many system modules.

Does the issue occur on stock atmosphere without custom system modules?

EggShweg, 2021-03-10:
I haven’t tested that. The only ones I use are tesla, sys-screenuploader, sys-con, and sys-clk. How would I get rid of the others??

EggShweg, 2021-03-10:
Edizon is most probably outdated

SciresM, 2021-03-10:
rename the atmosphere folder or delete it, then re-extract latest release zip to get a clean ams distribution.

EggShweg, 2021-03-10:
Could I then copy the old atmosphere folder over and skip files already on the sd card?

SciresM, 2021-03-10:
no.

EggShweg, 2021-03-11:
Closed because issue has been solved. Thanks a million!

closed 2021-03-11 by EggShweg