Atmosphere-NX/Atmosphere #1402: FW 11.0.1, Atmosphère 0.18.1, Title ID: 0100000000000002b / Error desc: std::abort() called (0xffe) on booting emuNAND

bejamartins, 2021-03-16:
## Bug Report

</br>
</br>

### What's the issue you encountered?

When I try to boot emuNAND (as files) I get the error:
"A fatal error ocurred when running Atmosphère.
Title ID: 0100000000000002b
Error desc: std::abort() called (0xffe)"

### How can the issue be reproduced?

Just trying to boot emuNAND.

### Crash Report
I don't have a recent crash report, only a binary file generated on atmosphere\fatal_errors: https://gist.github.com/bejamartins/a9b2e600a346620c74ff98360870e26f

### System Firmware Version

11.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate + Fusèe Primary
- Official release or unofficial build: 
  -Official release version 0.18.1
- Do you have additional kips or sysmodules you're loading: "Hekate+AMS-fss0-sigpatches-18.1-cfw-11.0.1"
- Homebrew software installed: Tinfoil

SciresM, 2021-03-16:
Piracy is not supported.

Your report is completely gibberish, because you uploaded it as a text file. 

Upload it as a binary file, as providing support for your issue is otherwise impossible.

closed 2021-03-16 by SciresM

bejamartins, 2021-03-16:
@SciresM Sorry, my bad. Updated the gist with the binary,

SciresM, 2021-03-16:
You have a corrupt PRODINFO/system save.

Follow the instructions/directions in https://github.com/Atmosphere-NX/Atmosphere/issues/1256 for guidance...I'm really not sure how people end up corrupting PRODINFO in this way, but yeah you'll need to restore a non-corrupt backup and/or delete the system settings savedata.

bejamartins, 2021-03-16:
@SciresM Thank you. For what it is worth what I did in the moment it first crashed and gave the error was connecting a pro controller to the dock via USB.