Atmosphere-NX/Atmosphere #1409: Atmos not booting after updating to 0.18.1

LeyendaV, 2021-03-22:
## Bug Report

### What's the issue you encountered?

I've been using Atmos since May, last year. Never had any single issue with any particular update so far, as long as I can remember.
Now, after updating to 0.18.1, Atmos crashes right after the Nintendo Switch splash logo appears on screen.

### How can the issue be reproduced?

Putting the Atmos M.18.1 files on the root of your SD card and booting it through Hekate.

### System Firmware Version

11.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: Official M.18.1
- Do you have additional kips or sysmodules you're loading: kips no, sysmodules yes.
- Homebrew software installed: Super Mario 64.

### Additional context?

- Additional info about your environment: Here's a picture of the crash screen.
[image]

Masamune3210, 2021-03-22:
what system modules do you have installed and what versions

LeyendaV, 2021-03-22:
> what system modules do you have installed and what versions

(00FF747765616BFF) sys-tweak
(0000000000534C56) SaltySD
(420000000007E51A) Tesla
(0100000000000352) Emuiibo

Latest release for each one.

Masamune3210, 2021-03-22:
Try a fresh install. No patches or sysmods other than what comes with atmos stock

SciresM, 2021-03-22:
Yes, please test without custom system modules to see if that resolves the problem.

I am dubious that SaltySD actually supports 11.0.x, in particular.

LeyendaV, 2021-03-22:
I've been using 11.0.x for a few months now and without any issue. Let me test a clean install, I'll report back.

closed 2021-03-22 by LeyendaV

masagrator, 2021-03-22:
SaltyNX supports 11.0.1.

Issue is that SaltyNX and emuiibo doesn't like each other, so using them both at the same time results in random errors.