Atmosphere-NX/Atmosphere #1414: 12.0.0 Support

SciresM, 2021-04-06:
Things to do for atmosphere to support 12.0.0.

* [ ] Reverse engineer new IPC protocol.
* [ ] Reverse engineer new kernel changes.
* [ ] Misc. Reverse engineering
* [ ] Implement new IPC protocol for SM.
* [ ] Implement new IPC protocol for pgl.
* [ ] Implement new kernel changes.
* [ ] Implement new misc. changes
* [ ] Implement client-support for new IPC protocol in libnx.
* [ ] Update secure monitor/libexosphere with new DRAM IDs.
* [ ] Update emummc patches to support new FS kips.
* [ ] Update nogc patches for new FS kips.
* [ ] Implement new kernel patches for mitm support (last time I'm doing this, because meso will be always-on soon).
* [ ] Update version numbers and other minor changes.
* [ ] Several issues were fixed, and usability and stability were improved.

jack420king, 2021-04-06:
what is the Fatal error: [NXBOOT]: Unable to identify package1! i have been geting when trying to run the fusee.bin from hekate

sapycola, 2021-04-06:
> what is the Fatal error: [NXBOOT]: Unable to identify package1! i have been geting when trying to run the fusee.bin from hekate

Hi, attend new version for support 12.0.0! stay tuned on sciresM software

Aonumiles, 2021-04-06:
Phew! I thought it was over since I updated my console without reading about the last version of Atmosphère not being compatible with the new firmware. I wont update without checking first in the future.
Thank you very much SciresM, hope you're doing great. :)

jack420king, 2021-04-06:
yeah i just started trying to cfw my switch got the hekate menu went to select fusee.bin and i got the error sucks so much missed  bye 1 day lol  danm auto update

splitlevelhead, 2021-04-06:
Hi please help. I have a impatient 4 year old. I updated mario Kart and now I get that dreaded eror message 
nintendo switch fatal error nxboot unable to identify package 1
I have formated card and put on fresh atmosphere 0.18.1 and Fusee payload but still no Joy.
What do I do??

Aonumiles, 2021-04-06:
> 
> 
> Hi please help. I have a impatient 4 year old. I updated mario Kart and now I get that dreaded eror message
> nintendo switch fatal error nxboot unable to identify package 1
> I have formated card and put on fresh atmosphere 0.18.1 and Fusee payload but still no Joy.
> What do I do??

Hi,

You'll have weather to reboot your Switch and launch without CFW or wait until the new version of Atmosphère release.

splitlevelhead, 2021-04-06:
sorry what do you mean by reboot your switch and launch without CFW?

jack420king, 2021-04-06:
how long dose it usually take for a new atmosphere release has any one been around that long

Aonumiles, 2021-04-06:
> 
> 
> sorry what do you mean by reboot your switch and launch without CFW

Hold the power button for like 10 seconds and turn the console on

splitlevelhead, 2021-04-06:
I only have copied games. Does that mean I have no choice to wait until new version of atmosphere? Or is there another way?
And how long could it take for a new version of atmosphere to come along??

Aonumiles, 2021-04-06:
> 
> 
> I only have copied games. Does that mean I have no choice to wait until new version of atmosphere? Or is there another way?
> And how long could it take for a new version of atmosphere to come along??

Just wait a few days, maybe until this week end

splitlevelhead, 2021-04-06:
ok . in future what do I do when it wants me to update a game? Coz if I dont click update it wont work??

closed 2021-04-11 by SciresM