Atmosphere-NX/Atmosphere #1418: Hi I Just bought a older switch

jonmcgovern, 2021-04-07:
I accidently updated today when there was a update and i just got this older switch today... and now I get a package1 error when trying to jailbreak my switch. It’s my fuse. I was wondering if there’s going to be a update soon for atmosphere ? 

Thanks 🙏🏻

closed 2021-04-07 by SciresM

jonmcgovern, 2021-04-07:
> I accidently updated today when there was a update and i just got this older switch today... and now I get a package1 error when trying to jailbreak my switch. It’s my fuse. I was wondering if there’s going to be a update soon for atmosphere ?
> 
> Thanks 🙏🏻

Why it get closed? I’m so confused .-.

SciresM, 2021-04-07:
There is an existing issue tracking TODOs for 12.0.0 support.

Additionally, you did not use the issue templates.

Masamune3210, 2021-04-07:
Because you didn't fill out the issue template, instead electing to delete it and ask the same question literally every one is asking right now. Wait for atmos to get a update for the new firmware

jonmcgovern, 2021-04-07:
I apologize I’m new to GitHub. Thank you