Atmosphere-NX/Atmosphere #1422: Will not Load Into and Certian Program Called ChoiDujourNX

TheDogMatt, 2021-04-11:
## Bug Report

Error Code: 2144-0001 (0x290)
Program: 0100000000001000
Firmware:12.0.0 (Atmosphere M 0.19.0-master-88dd4147)

### What's the issue you encountered?

I was trying to load into Choidujour after going through album and went to blue error screen
### How can the issue be reproduced?

### Crash Report
[01618146155_0100000000001000.log](https://github.com/Atmosphere-NX/Atmosphere/files/6291831/01618146155_0100000000001000.log)

### System Firmware Version

12.0.0
### Environment?

atmosphere - Latest

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2021-04-11:
Use daybreak instead.

closed 2021-04-11 by SciresM

TheDogMatt, 2021-04-11:
Ok i Will try that thank you