Atmosphere-NX/Atmosphere #1424: Game card crashes switch

mbamber1986, 2021-04-11:
Hi so I stupidly updated my system nand and it updated my game cart Fw as well 

Anyway since updating it doesn't work  with ams 18 and 19 emunand and crashes the switch 

I've applied nogc=0 but it's still doing the same 

I don't think this is a bug on your side but is my only option in order to use my cards on emunand to 12.0 or is there a fix

Tia

mgs04, 2021-04-11:
same thing for me

Nanolx, 2021-04-11:
What's the error message that atmosphere throws? Did you update fusee-primary aswell and/or hekate? For me it's working fine, both OFW and CFW at 12.0, Atmosphere 0.19.0[2] and hekate 5.5.5.

Also Atmosphere 0.19.0 got an updated build, do you use that build?

`Note: The NCM build in the 0.19.0 zip was updated on 2021/04/11 at 5:45 AM PST to fix a random crash when deleting games. The updated zip has a build hash of "-c67c29eb".`

SciresM, 2021-04-11:
If you have ever used gamecards on >=12.0.0, you can never use them on <12.0.0.

configure nogc appropriately.

closed 2021-04-11 by SciresM

mbamber1986, 2021-04-11:
Thanks @SciresM i.have an emunand and a sysnand and that's what I did thanks for your help I think my only option for now is wait till 12.0 is stable and upgrade 

Thanks for your input :)