Atmosphere-NX/Atmosphere #1426: upt 12.0.0 and AMS 0.19.0 dont work

LordElrond1903, 2021-04-11:
I updated my switch to 12.0.0
Then applied what you released
I download Hekate and Sigpatches.
But I cant make Tinfoil work or install my games.
I have some reports and gonna send them to you.
Thanks in advance for your great work.
[01618153993_0100000000001000.log](https://github.com/Atmosphere-NX/Atmosphere/files/6292158/01618153993_0100000000001000.log)
[01618154340_010000000000003a.log](https://github.com/Atmosphere-NX/Atmosphere/files/6292159/01618154340_010000000000003a.log)

mbamber1986, 2021-04-11:
I don't think there are any sigpatches out yet for 12.0.0 your best sticking with 11.0.1 till they are available

LordElrond1903, 2021-04-11:
Downgrade causes brick your console ?

mbamber1986, 2021-04-11:
Are you using emunand or sysnand

TheDogMatt, 2021-04-11:
> 
> 
> I don't think there are any sigpatches out yet for 12.0.0 your best sticking with 11.0.1 till they are available

There Are SigPatchs Out Already

mbamber1986, 2021-04-11:
Are there when did they release couldn't find them for 12.0.0

MonsieurKimYong, 2021-04-11:
When excute Non applet mod "Hbmenu", Error occured. Games are going well...but that cause....I can't use install app 'TinLeaf'

Using 19.0.0....

LordElrond1903, 2021-04-11:
> Are you using emunand or sysnand

Im a newbie about hacking. How can I understand it is emunand or sysnand ?

fincs, 2021-04-11:
Piracy is not supported.

closed 2021-04-11 by fincs