Atmosphere-NX/Atmosphere #1430: Atmosphere 19.0 Not Booting Properly Into emuMMC

Sam-Man16, 2021-04-11:
## Bug Report

### What's the issue you encountered?

 After updating OFW to 12.0 and dumping it, I attempted to update emuMMC using daybreak (atmosphere/hekate are both fully updated) after updating the firmware was complete I pressed reboot to see the atmosphere logo then a black screen. I attempted numerous times to get emuMMC up and running and succeeded twice. When the console is in the black screen state my computer still detects the console as on and connected, but no display or any other functions. </br>
 No changes were made to Atmosphere</br>

### How can the issue be reproduced?

 1. Attempt to update emuMMC firmware to 12.0 using daybreak.
  2. Reboot. </br>

### Crash Report

 Crash reports can be found under ``/atmosphere/crash_reports``. </br>
 If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. </br>
 If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. </br> 
No crashes occured.

### System Firmware Version

12.0.X</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate v5.5.5
- Official release or unofficial build: 
  -  Official release version 0.19.x 
- Do you have additional kips or sysmodules you're loading: sys-clk (not active at the time)
- Homebrew software installed: [ daybreak.nro, fakenews-injector.nro, reboot_to_payload.nro, retroarch_switch.nro, sys-clk-manager.nro, vba-next.nro ]

### Additional context?

- Additional info about your environment:
-  I am not sure but it seems like the emuMMC still boots, no inputs work but my computer still detects the nintendo switch via USB cable. Holding the power button for 12 seconds disconnects it. Injecting Fusee-Primary produces the same results. I did get the console to boot into emuMMC twice with no issues but all other times the aforementioned issue occurs.

SciresM, 2021-04-11:
"not active at the time" this is not how system modules work.

Does the error occur with nothing at all inside /atmosphere/contents? (Delete the directory, to test).

Sam-Man16, 2021-04-11:
> "not active at the time" this is not how system modules work.
> 
> Does the error occur with nothing at all inside /atmosphere/contents? (Delete the directory, to test).

Sorry what I meant is it was not overclocking anything at that time. I will test this now.

Sam-Man16, 2021-04-11:
> "not active at the time" this is not how system modules work.
> 
> Does the error occur with nothing at all inside /atmosphere/contents? (Delete the directory, to test).
After three trial runs it appeared to have fixed the issue, thank you. I will attach the contents folder below which was on the SD during the issue. Please forgive any ignorance in this matter, I am not well versed when it comes to the homebrew scene.

[contents.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6292416/contents.zip)

WeeabooGod, 2021-04-11:
This is actually related to Hekate. After Updating Hekate and Atmosphere I found myself unable to load into anything. I realized I had to update my sig patches, which finally allowed me to go into my CFW, and loading fuse-primary as a payload would bring me into EMUMMC Correctly. 

I didn't save the hekate_ipl file from before but I remember removing some unnecessary checks.
why say atsmopshere = 1 when ffs0 has it enabled automatically. If I need to emummc_force_disable for SysNand CFW then why do I need to use emummc = 1. it was that value that caused  the strange issue.

Its probably not the main cause, but removing it allowed me to load back into EmuMMC.

Sam-Man16, 2021-04-11:
> Does the error occur with nothing at all inside /atmosphere/contents? (Delete the directory, to test).
What would cause the contents folder to stop atmosphere from booting? If I boot without contents it effectively renders the point of using emuMMC useless.

SciresM, 2021-04-11:
Custom system modules hanging the boot process because they need to be updated. 

Not sure at all why you think it's pointless without that folder?

closed 2021-04-11 by SciresM

Sam-Man16, 2021-04-11:
> Custom system modules hanging the boot process because they need to be updated.
> 
> Not sure at all why you think it's pointless without that folder?

None of my CFW titles run without the folder.

SciresM, 2021-04-11:
Any titles in that folder are non-atmosphere; they'll just need to be updated on a case by case basis by their developers.