Atmosphere-NX/Atmosphere #1431: Monster Hunter Rise DLC content Crash

RushChill, 2021-04-11:
**What's the issue you encountered?**

After updating Atmosphere to 19.0, I opened MonHun Rise and when the program starts loading the DLC content it crashes.

**How can the issue be reproduced?**

Loading the Monster Hunter Rise game (digital version) with DLC content

Crash Report

Nintendo System Error Message: The Software Was Closed Because An Error Occurred

**System Firmware Version**
12.0.X

**Environment?**
    What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate v5.5.5
    Official release or unofficial build:
    Official release version 0.19.x
    Do you have additional kips or sysmodules you're loading: No (eliminated the atmosphere/contents folder before turning on the system
    Homebrew software installed: [ daybreak, reboot_to_payload, kosmos_toolbox,  checkpoint, ftspd_pro, ftspd_classic, jksv, all_in_one_update ]
   Sigpatches updated for 12.0

**Additional context?**
    After updating to 19.0, I cannot enter to emuMMC, only via fusee_primary

SciresM, 2021-04-11:
Checked for corrupted data? Redownloaded the dlc from the eshop?

RushChill, 2021-04-11:
> Checked for corrupted data? Redownloaded the dlc from the eshop?

Yeah, already did both and nothing, it crashes in the same load screen

urherenow, 2021-04-12:
Whether nor not your content is legit, do you have reason to have FS/ES patches on your system? I don't know the details, but the latest patches were released like 4 or 5 times last night. If you have any of those on your system, I would try to re-download and install them again to see if it takes care of your issue here.

masagrator, 2021-04-12:
> Sigpatches updated for 12.0

Piracy is not supported.

SciresM, 2021-04-12:
Can the generated crash report be uploaded, please?

SciresM, 2021-04-13:
Closing, because this is probably the content enumeration issue I just fixed in 0.19.1.

If the error continues to occur with 0.19.1, please let me know, and I'll try to get someone to reproduce it.

closed 2021-04-13 by SciresM