Atmosphere-NX/Atmosphere #1433: Error 2002-2007(0xe02) When moving Pokemon files saves between devices

CarlosRamirezGriego, 2021-04-11:
## Bug Report

### What's the issue you encountered?

I get an error screen with Error 2002-2007(0xe02)
Program: 010000000000003A
Firmware: 12.0.0
Atmosphere: M 0.19.0-master-88dd4147

### How can the issue be reproduced?

I have 2 original cartridges for Pokemon Shield
1 cartridge is in the CFW switch
1 cartridge is in the Non-CFW switch
When I try to move the "Save" from the Non-CFW to the CFW switch, I get this error

Note: I have been doing this for a while, so something from the recent release broke this feature
Note: This also happens when I start FTPD 2.2.0, havent tried other apps

### Crash Report

[crash_reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6292492/crash_reports.zip)

### System Firmware Version

12.0.0|AMS M.19.0|E

### Environment?

- Bootloader: Hekate 5.3.2
- Do you have additional kips or sysmodules you're loading: Not that I know
- Homebrew software installed: Checkpoint, FTPD 2.2.0, CaptureSight, Daybreak, Edizon SE, Goldleaf, HB App Store, Kosmos Tool, Lockpick, Tinfoil

### Additional context?

[image]

SciresM, 2021-04-11:
Yeah migration is double launched right now I think, and so will be broken.

Already committed a fix, but hadn't done a shadow release. Kinda leaning towards telling you to wait for 0.19.1 as more issues potentially get fixed in the short term future?

CarlosRamirezGriego, 2021-04-11:
> Yeah migration is double launched right now I think, and so will be broken.
> 
> Already committed a fix, but hadn't done a shadow release. Kinda leaning towards telling you to wait for 0.19.1 as more issues potentially get fixed in the short term future?

Its alright, I will wait for the 0.19.1 release, thanks!

Yakimurai, 2021-04-11:
> Yeah migration is double launched right now I think, and so will be broken.
> 
> Already committed a fix, but hadn't done a shadow release. Kinda leaning towards telling you to wait for 0.19.1 as more issues potentially get fixed in the short term future?

Thank you so much for your work

SciresM, 2021-04-13:
Closing, since 0.19.1 should have the migration double launch fixed.

closed 2021-04-13 by SciresM