Atmosphere-NX/Atmosphere #1437: Error 2001-0123 (0xf601) Atmosphere crashes upon boot

jazz-g, 2021-04-11:
Error Code 2001-0123 (0xf601)
Program: 010000000000001F
Firmware: 12.0.0 (Atmosphère M 0.19.0-master-88dd4147)
Bootloader: Hekate 5.5.5
Did clean install of Hekate and Atmosphère, completely deleting and replacing /sept, /switch, /hbmenu.nro, and /atmosphere

Atmosphere logo sequence shows upon boot, then Nintendo's logo, then it crashes. Stock switch works fine.
Help would be much appreciated

Error Screen
[image]

The-NinToaster, 2021-04-11:
Having the same issue. Exact same conditions as OP. Following.

SciresM, 2021-04-11:
Will be looking into this soon. Gotten two reports of it.

The-NinToaster, 2021-04-11:
> 
> 
> Will be looking into this soon. Gotten two reports of it.

Keep up the good work, brother.

ryanl85, 2021-04-12:
same issue as op, it boots as far as the lock screen then crashes.

Skonikol, 2021-04-12:
You need to take out the game card

ryanl85, 2021-04-12:
dont have one in. same issue.

ryanl85, 2021-04-12:
[image]
this is the error i get.

Skonikol, 2021-04-12:
> [image]
> this is the error i get.

This is a different problem. You need to remove the ldn_mitm module

ryanl85, 2021-04-12:
that fixed it. thx!!!!!!!!!

Walker-HK, 2021-04-12:
same issue

polizie, 2021-04-12:
same issue error code 2001-0123(0xf601)
program:010000000000001f

switch lite modchip

jesusrh2896, 2021-04-12:
Solo eliminen la carpeta de contents, que está dentro de la carpeta de atmosphere

jazz-g, 2021-04-12:
I removed the gamecard, the same issue persists. I have a clean atmosphere + hekate install. Should I reformat the card and try again?

SciresM, 2021-04-12:
This issue is caused by a partial update being applied to the console, where 12.0.0 NCAs are installed but BOOT0 still contains 11.0.0 (or earlier).

This causes fusee to report target firmware = 11.x, and so PGL uses CMIF, but NS tries to use TIPC to communicate.

The solution is for affected users to manually apply the BOOT0 update by writing correct BCT/Package1, and/or Package2 (since if pk11 wasn't written pk21 probably wasn't).

I will be looking into how this is even possible, though -- I don't see how it can happen unless the console is shut down during an update or something? 

This may be related to reports of a bug in NCM regarding ListContentId.

Closing this issue for now, since it's solved, hopefully whatever the ListContentId bug is resolves neatly.

closed 2021-04-12 by SciresM

Vanxhi01, 2021-04-13:
> This issue is caused by a partial update being applied to the console, where 12.0.0 NCAs are installed but BOOT0 still contains 11.0.0 (or earlier).
>
> This causes fusee to report target firmware = [...]

I know this issues have been closed. How and where exactly do I need to manually apply the BOOT0?

jazz-g, 2021-04-13:
How I'm fixing it is by doing the firmware update again, by restoring a raw NAND backup so that the firmware version matches boot, (no fuses were blown in 12.0.0), then re-updating the firmware. I'm still fairly new, but I think this should work, someone smarter than me let me know if I will break everything by doing this

SciresM, 2021-04-13:
I also believe that should work.

If it doesn't, contact me on Discord, and I'll help you get it sorted.

Jorlogamer, 2021-04-14:
I have the same error how can i update boot0?

Masamune3210, 2021-04-14:
Read literally the last two messages right above yours.....

TheKidThatCodes, 2023-03-17:
how to do this without a nand backup
my nand backup is from 15.something and i updated to 16.0.0 without using hombrew, so i think fuses are blown.
im asking because i want to make new emunands because i have this issue on my 2 current ones and on my sysnand

DarkSkull91, 2023-06-10:
Hello guys, I have this same issue trying to update to 16.0.3 firmware and Atmosphère 1.5.3. How can I solve this?

lurondev, 2023-06-14:
Same issue with 16.0.3 and Atmosphère 1.5.3. Any solutions?

chronz1, 2023-06-14:
> Same issue with 16.0.3 and Atmosphère 1.5.3. Any solutions?

Same here, Im updating from 13.1.0 to the newest firmware 16.0.3 and I updated my AMS to 1.5.3/Hekate 6.0.4

mhussle, 2023-06-14:
> 

I am literally in the same boat.

DarkSkull91, 2023-06-14:
I'm not at home but I solved it updating also hbmenu or similar... Try to upgrade all your applications.

chronz1, 2023-06-14:
OK , i "solved" it by deleting the folder associated with the fatal report. 

"420000000007e51a"

This was the folder the internet suggested, that aligned with my report as well.

cypira, 2023-06-14:
> OK , i "solved" it by deleting the folder associated with the fatal report.
> 
> "420000000007e51a"
> 
> This was the folder the internet suggested, that aligned with my report as well.

After updating sysmmc and rebooting, I had the same error.

Following the above from chronz1, I can confirm I have carried out the same action and cfw sysmmc now boots for me. I do worry though if the error I too have now "fixed" is suggestive of a larger problem. I had a failed update in emummc so, yeah good times!

horst789github, 2023-06-14:
> 

this works for me

Masamune3210, 2023-06-16:
This is why one of the first steps to troubleshooting issues is to remove all sysmodules. 7e51a is Tesla, its likely either out of date or conflicting with another systemodule you have.

EVANGELION-K, 2023-07-29:
> 感谢这对我有用,所以我猜测是第三方插件例如特斯拉,与系统不兼容所导致的.

204870, 2023-09-04:
just came in to say that this solution worked for me too!

> OK , i "solved" it by deleting the folder associated with the fatal report.
> 
> "420000000007e51a"
> 
> This was the folder the internet suggested, that aligned with my report as well.

Masamune3210, 2023-09-05:
Guys, we get it, deleting the out of date custom system module does in fact fix the issue caused by the out of date system module. Can we please stop commenting on the closed issue?