Atmosphere-NX/Atmosphere #1443: to

fmg3656, 2021-04-13:
## Bug Report

### What's the issue you encountered?
When updating to Atmosphere 0.19.1 on FW 12.0.0 from Atmosphere 0.19.0, I get a fatal error and a log dump.
### How can the issue be reproduced?
It can be reproduced by downgrading to Atmosphere 0.19.0 and trying to reinstall 0.19.1 

### Crash Report
Atmosphère Fatal Report (v1.1):
Result:                          0x115D (2349-0008)

Program ID:                      010000000000100d
Process Name:                    hbloader
Firmware:                        11.0.1 (Atmosphère 0.18.1-master-1545fa9d)
General Purpose Registers:
Start Address:                   0000000000000000
Stack Trace:
        ReturnAddress[00]:       0000000000000000
        ReturnAddress[01]:       0000000000000000
        ReturnAddress[02]:       0000000000000000
        ReturnAddress[03]:       0000000000000000
        ReturnAddress[04]:       0000000000000000
        ReturnAddress[05]:       0000000000000000
        ReturnAddress[06]:       0000000000000000
        ReturnAddress[07]:       0000000000000000
        ReturnAddress[08]:       0000000000000000
        ReturnAddress[09]:       0000000000000000
        ReturnAddress[10]:       0000000000000000
        ReturnAddress[11]:       0000000000000000
        ReturnAddress[12]:       0000000000000000
        ReturnAddress[13]:       0000000000000000
        ReturnAddress[14]:       0000000000000000
        ReturnAddress[15]:       0000000000000000
        ReturnAddress[16]:       0000000000000000
        ReturnAddress[17]:       0000000000000000
        ReturnAddress[18]:       0000000000000000
        ReturnAddress[19]:       0000000000000000
        ReturnAddress[20]:       0000000000000000
        ReturnAddress[21]:       0000000000000000
        ReturnAddress[22]:       0000000000000000
        ReturnAddress[23]:       0000000000000000
        ReturnAddress[24]:       0000000000000000
        ReturnAddress[25]:       0000000000000000
        ReturnAddress[26]:       0000000000000000
        ReturnAddress[27]:       0000000000000000
        ReturnAddress[28]:       0000000000000000
        ReturnAddress[29]:       0000000000000000
        ReturnAddress[30]:       0000000000000000
        ReturnAddress[31]:       0000000000000000

### System Firmware Version
12.0.0

### Environment?

Hekate 5.5.5 bootloader
Official release: Atmosphere 0.19.1 
  - Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: None

### Additional context?
None
[01618351468_010000000000100d.log](https://github.com/Atmosphere-NX/Atmosphere/files/6307148/01618351468_010000000000100d.log)
[01618351468_010000000000100d.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6307149/01618351468_010000000000100d.zip)

SciresM, 2021-04-13:
"100D" is a hbl error. 

This indicates an error in some homebrew.

I don't understand what you mean, or how it's even possible to get a hbl error "while updating atmosphere".

Atmosphere cannot be updated on device; it can only be updated by copying files to the SD card while the console is not running.

I don't think this is an atmosphere issue.

SciresM, 2021-04-13:
The error you posted is also from 0.18.1, not 0.19.0 or 0.19.1.

closed 2021-04-13 by SciresM

fmg3656, 2021-04-13:
I know thats what it says, but I assure you it is while copying files from 0.19.0 to 0.19.1.  Nothing else has been changed on the system

SciresM, 2021-04-13:
"while copying files from 0.19.0 to 0.19.1" but you can't update atmosphere on device. This sounds like a user error.

fmg3656, 2021-04-13:
No, I was updating on the PC... then inserting the card into the switch.  upon reboot I get this error.

SciresM, 2021-04-13:
Take a picture of the error you get. I guarantee it is not the error you uploaded.

fmg3656, 2021-04-13:
I recently upgraded to atmosphere from sxos.  I went to the switch folder and deleted all of the sx programs and the error went away