Atmosphere-NX/Atmosphere #1444: Crash when Opening Web Applet through Super Mario Odyssey

SurprisedPikaa, 2021-04-15:
## Bug Report

### What's the issue you encountered?

When opening the action guide in Super Mario Odyssey, which uses the web applet, I am experiencing a fatal error and a system crash.<br>
Atmosphere was unmodified with only a few sysmods running that shouldn't be the cause of the issue</br>

### How can the issue be reproduced?
</br>Open up Super Mario Odyssey. I was using the digital 1.3.0 version, but I'm unsure if it works/does not work on other versions. In the main menu, select the "action guide" option to open it or create a new file and in the cutscenes between the Cascade Kingdom and the Sand Kingdom the action guide will be automatically opened. Around 50% of the time this will result in an error and the console will immediately crash.</br>

### Crash Report
No crash report seems to have been generated, I'm not sure why. I've attached an image of the crash below.</br>
[image]

### System Firmware Version
12.0.0</br>
### Environment?

- Atmosphere was launched by Hekate using hekate_ctcaer_5.5.5.bin.
- Official release version 0.19.1
- Homebrew software installed: SysDVR, IPSwitch, HB Appstore, all homebrew related to deepsea (all in one updater, deepsea toolbox)

### Additional context?
N/A

SciresM, 2021-04-15:
Does the error occur without custom system modules?

Also, the image you posted describes the location of a fatal error binary. Please upload that binary.

SurprisedPikaa, 2021-04-15:
Here is the file you asked for. Error doesn't seem to happen without custom system modules, but the error doesn't happen 100% of the time so it could be just me getting lucky.
[report_34f497e8af.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6317324/report_34f497e8af.zip)

SurprisedPikaa, 2021-04-15:
Correction: it has just happened without any sys modules.

SciresM, 2021-04-15:
Can you post the report that occurred with no custom system modules, and a screenshot of your accompanying /atmosphere/contents folder?

The report you posted is *really* weird.

It's a crash in the crash handler, so something else must be going wrong elsewhere for this to even happen in the first place.

SurprisedPikaa, 2021-04-15:
[image]
This is my contents folder. When testing without sysmods I temporarily deleted the contents of it. 
[report_a9c7dcd66.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6317656/report_a9c7dcd66.zip)
Here is the report without sysmods.

SciresM, 2021-04-15:
You fully deleted the contents of that folder when testing no-sysmodules?

That report still seems completely nonsense...it describes a failure to connect to `ams:bpc` to do a fatal reboot.

but a) for this code to be executed there must already be a fatal happening
b) how can a fatal succeed after one has failed?

I guess it could be because of the limit increases I hadn't properly imported...please put the attached .kip file in the zip inside /atmosphere/kips.

Does the error continue to occur, if you do that?

Send a report with that file inside /atmosphere/kips?

Please note that if you use hekate and not fusee, you'll need to make sure that you have `kip1=atmosphere/kips/*` in your hekate_ipl boot entry.

[pm.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6317722/pm.zip)

SurprisedPikaa, 2021-04-15:
I don't really understand what "Please note that if you use hekate and not fusee, you'll need to make sure that you have kip1=atmosphere/kips/* in your hekate_ipl boot entry." means, could you dumb that down for me lol

SciresM, 2021-04-15:
Hekate doesn't automatically load all KIPs inside /atmosphere/kips...possible for you to test booting with fusee-primary instead of hekate, to make stuff simpler?

Shouldn't affect whether the bug occurs, if you want to test using hekate you'll need to edit your boot configuration file in a way that's a little complicated to explain over GitHub comments.

SurprisedPikaa, 2021-04-15:
> You fully deleted the contents of that folder when testing no-sysmodules?
Yeah I just copied it over to my computer.
> possible for you to test booting with fusee-primary instead of hekate
Ok I will try that now. I don't have a folder at atmosphere/kips/ , should I create it or use atmosphere/kip_patches ?

SciresM, 2021-04-15:
Create it. the pm.kip file I sent should be taken out of the zip and put inside `/atmosphere/kips`, not kip_patches.

SurprisedPikaa, 2021-04-15:
I can't seem to recreate the error using fusee-primary and the file you sent.

SciresM, 2021-04-15:
Does the error occur with fusee primary but the kip I sent not on SD?

If so, I'll assume the error is that I forgot to update the resource limits in PM, for which a fix was already committed to master (and will be included in 0.19.2 in a week or two).

SurprisedPikaa, 2021-04-15:
Yes the issue happens with fusee-primary but without the kip you sent

SciresM, 2021-04-15:
Yeah, okay, that resolves it.

closed 2021-04-15 by SciresM

FisherMario, 2023-02-25:
Hi, 

I've got the same problem now, but on Atmosphere 1.4.1, each time I try to open the guide, a fatal error report shows up. I have attached the report. 

What I've tried:
- Update Atmosphere and Hekate (I had the same problem on Atmosphere 1.4.0).
- Update SIG patches
- Reinstall Mario Odyssey (both with DBI as with Goldleaf)
- Remove the game, ticket, saved games etc. 
- Read the messages above, but in this Atmosphere version the KIP folder is named different. 

I do not use any cheat codes or custom theme or anything (I never did). 

Do you have any idea what can be wrong?

Thanks in advance!

[01677310735_010000000000100f.log](https://github.com/Atmosphere-NX/Atmosphere/files/10832267/01677310735_010000000000100f.log)

SciresM, 2023-02-25:
Piracy is not supported.

tImMiTeNeRgY, 2024-01-07:
I have the Same issue :(

masagrator, 2024-01-07:
> I have the Same issue :(

Are you on the latest Atmosphere version?

tImMiTeNeRgY, 2024-01-07:
yes, I downloaded it earlier

tImMiTeNeRgY, 2024-01-07:
> ![20240107_135717](https://private-user-images.githubusercontent.com/124054214/294766593-6adcedbe-5674-4a5a-851d-34512df097ab.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2NDAyNzUsIm5iZiI6MTcwNDYzOTk3NSwicGF0aCI6Ii8xMjQwNTQyMTQvMjk0NzY2NTkzLTZhZGNlZGJlLTU2NzQtNGE1YS04NTFkLTM0NTEyZGYwOTdhYi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwN1QxNTA2MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yZTUzODdmMjU3ZWEzZTk1ODVlY2E4NTMyNjE5ZmVkZWM5MjlmNDc5YTgxNzY2MjRmZWE5MmFlM2JjYjczNWZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ZK31eBIVHUaGJniJHhu24ftG6eXK5zJg_3u9SO5_SHU) And this is the error

Ah, this is a picture before i updated the version

tImMiTeNeRgY, 2024-01-07:
[image]
This is with latest version

tImMiTeNeRgY, 2024-01-08:
I fixes it