Atmosphere-NX/Atmosphere #1445: Error Code 2144-0001

nomo9133, 2021-04-16:
When I run any .nro(ex: ChoiDujourNX,checkpoint ) program, the system will show error code 2144-0001
but when I use EmuMMC(Firmware: 9.0.1(Atmosphère 0.17.1)) ,it is normal.
how can i fix the error?
thanks

```
Atmosphère Fatal Report (v1.1):
Result:                          0x290 (2144-0001)

Program ID:                      0100000000001000
Process Name:                    qlaunch
Firmware:                        12.0.0 (Atmosphère 0.19.1-master-997e4dd6)
```

drofdoom, 2021-04-16:
Daybreak and edison .

SciresM, 2021-04-16:
1) you did not use the issue templates.

2) certain homebrew needs to be updated to be used post 12.0.0 or ams 0.19.0, due to sm changes.

For checkpoint: https://github.com/SciresM/Checkpoint/releases/tag/ams-19-compat

For choi: don't use it, use daybreak instead.

closed 2021-04-16 by SciresM

nomo9133, 2021-04-16:
> Daybreak and edison .

thanks :) have a nice day