Atmosphere-NX/Atmosphere #1446: panic occured while running atmosphere. title ID: 010000000000002B

macintoshprime, 2021-04-16:
## Bug Report

hello everybody, just for context Im pretty new with the atmosphere and so I got pretty shoocked with this error but here we go 

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
The error just started when I pulled my sd card accidentaly with the console on rest mode, thinking it was powered off (dumb mistake) and now i believe some archive may be corrupted 
N/A
### How can the issue be reproduced?

1-boot switch 
2- open the emunand
3- open any game and runs it without major problems 
4- when you closes the game the error happens, showing this on screnn: 

panic occured while running atmosphere
title ID: 010000000000002B

it has happened to all games that i tested and sometimes it doesn´t let the switch boot up 

### Crash Report

https://gist.github.com/macintoshprime/7209b3418ab1387213e5c4b0dae6120f

i extracted this error from /atmosphere/fatal reports

### System Firmware Version

10.0.2

Masamune3210, 2021-04-16:
Check filesystem for errors, rextract atmostphere to sd card, make sure system is off before removing storage

macintoshprime, 2021-04-16:
Should I remove /atmosphere and paste it again in the sd root?

macintoshprime, 2021-04-16:
> Check filesystem for errors, rextract atmostphere to sd card, make sure system is off before removing storage

Should I remove /atmosphere and paste it again in the sd root?

Masamune3210, 2021-04-16:
Yeah, at least move it off of the sd card, that way you don't lose your configs if any are modified

SciresM, 2021-04-16:
You did not upload the correct crash report. The filename will be shown when an error occurs, the file will be inside /atmosphere/fatal_errors.

macintoshprime, 2021-04-16:
> You did not upload the correct crash report. The filename will be shown when an error occurs, the file will be inside /atmosphere/fatal_errors.

inside this paste of my switch theres lots of files .BIN that i cant open, all of them titled out "report_" and the a code

SciresM, 2021-04-16:
Trigger the crash.

the filename will be shown on the screen.

upload that specific report.

SciresM, 2021-04-16:
The file will be a .bin, yes. I can open it, when you upload it.

macintoshprime, 2021-04-16:
sorry for the question but how do I upload here a .BIN file? im trying to upload it as a .zip file but still cant

SciresM, 2021-04-16:
Upload it to whatever file host you want.

macintoshprime, 2021-04-16:
http://www.filehost.pt/2MfuY
--

got it, should be here

SciresM, 2021-04-16:
...okay, please use a less shitty file host, that doesn't make me wait a minute and then "fail to generate download links".

Use https://up1.secretalgorithm.com/ or something.

macintoshprime, 2021-04-16:
https://up1.secretalgorithm.com/#9QtgncLTMhQD4cDmKDDrmg 

done, sorry for anything im just not that into technology hahah

SciresM, 2021-04-16:
1) You appear to be using atmosphere 0.18.1, which is outdated, instead of 0.19.1.
2) The error is "fs::ResultInvalidFatFormatSd()" -- your SD card's FAT filesystem is corrupted.

Reformat your SD card as FAT32 or switch to a new SD card, and the issue should be resolved.

closed 2021-04-16 by SciresM

macintoshprime, 2021-04-16:
thx, just going to do that

macintoshprime, 2021-04-17:
How do I 

> 1. You appear to be using atmosphere 0.18.1, which is outdated, instead of 0.19.1.
> 2. The error is "fs::ResultInvalidFatFormatSd()" -- your SD card's FAT filesystem is corrupted.
> 
> Reformat your SD card as FAT32 or switch to a new SD card, and the issue should be resolved.
How do I format my sd card without losing my nand informations? Should I backup the file of emummc?

cheesycod, 2021-04-17:
What exactly do you mean by loosing your nand information, do you mean your enummc, then yes, you should probably just backup your whole SD card and reformat and copy back the files (though atmosphere itself may likely be corrupted so redownload that)

macintoshprime, 2021-04-17:
> What exactly do you mean by loosing your nand information, do you mean your enummc, then yes, you should probably just backup your whole SD card and reformat and copy back the files (though atmosphere itself may likely be corrupted so redownload that)

I mean like when formating my sd somewhat it causes my console to brick. 
I've heard a few people talking about this and the i got a little afraid because I don't know what causes it