Atmosphere-NX/Atmosphere #1447: Error 2002-6605 when trying to use forwarders since 0.19 update

LightShaft, 2021-04-16:
### What's the issue you encountered?

Before 0.19 update, I could use forwarders just fine (for retroarch games, or Sonic AIR or Tinfoil or whatever) but now it doesn't and it crashes immediately. Error code : 2002-6605, program : 0100000000000023. I'm still on firmware 11.0.1. Obviously I've tried to update sigpatches and all. I've tried booting with Hekate fusee secondary and fusee primary, it doesn't change anything. I've tried to completely delete atmosphere's folder to start fresh, doesn't change. Posting here is my last resort.</br>

### How can the issue be reproduced?

Just launching a forwarder, doesn't matter which one.</br>

### Crash Report

[01618583149_0100000000000023.log](https://github.com/Atmosphere-NX/Atmosphere/files/6325873/01618583149_0100000000000023.log)

### System Firmware Version

11.0.1</br>

### Environment?

Fusée primary but I also tried with Hekate Fusee Secondary.
Official release of 0.19 and 0.19.1
Tried with my setup with sysmodules like sysclk teslamenu and fizeau running and with a fresh installation.

masagrator, 2021-04-16:
Using sigpatches is admitting to piracy.
Piracy is not supported.

Only supported way then and now is using title replacement mode for accessing full RAM. Forwarders were never supported by Atmosphere and never will be.

fincs, 2021-04-16:
Piracy is not supported.

Switch homebrew applications are NRO files, loaded through nx-hbloader, which in turn is loaded through Atmosphère's native support for it. Any other claimed solution for distributing or loading homebrew is bogus, and totally unsupported. For what's worth, "forwarders" are technologically indistinguishable from pirated games, and they are treated as such by both Nintendo and us.

closed 2021-04-16 by fincs