Atmosphere-NX/Atmosphere #1448: BRING BACK MODDING SUPORT

mjcox244, 2021-04-16:
## Feature Request

[ Please search for existing [feature requests](https://github.com/Atmosphere-NX/Atmosphere/issues?utf8=%E2%9C%93&q=is%3Aissue+label%3A%22features%2Ffeature-request%22) before you make your own request. ]</br>
[ Duplicate requests will be marked as such and you will be referred to the original request. ]

### What feature are you suggesting?
#### Overview:
The latest atmosphere limits/prevents game modding

#### Smaller Details: 
Due to the latest version of atmosphere includeing

> Atmosphère no longer uses the /contents/ folder for its own programs.
> Atmosphère's system modules are now bundled together in the single file "stratosphere.romfs".
> For those working on developing for atmosphère, executables inside the /contents/ directory will be preferred to those in "stratosphere.romfs".
> Please Note: In order to facilitate this change (and the desired behavior), the first time you boot after extracting a release zip, atmosphère system modules inside /contents/ will be deleted.
> This will have no impact on user programs (it only removes programs with specific program ids).

This prevents romfs modding that used this contents folder

#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Change
BRING THIS CONTENT FOLDER BACK FOR MODDING

### What component do you feel this would best fit within?
IDK whatever handled that contents foldeer

### Why would this feature be useful?
This will allow people to use the many romfs mods avalible without haveing to resort to an old version of atmosphere or editing NSP/XCI files

pixel-stuck, 2021-04-16:
you realize that you can still use the contents folder, correct? Just that the sysmodules that atmo ships no longer reside there. This is all the changelog says.

Conan179, 2021-04-16:
@mjcox244 Which mod shouldn't go?

mjcox244, 2021-04-16:
ah I was confused for a second as someone on the 3D world modding server was saying it no longer worked for their mods then noticed that

mjcox244, 2021-04-16:
NVM false alarm the person in the 3D world modding server just didn't lay the mod out correct

mjcox244, 2021-04-16:
I feal stupid now lul

closed 2021-04-16 by mjcox244

SciresM, 2021-04-16:
Do not create all caps issue please.